azlux/log2ram #10: It is really working?

hrvstr, 2017-02-27:
The [README](https://github.com/azlux/log2ram#it-is-working-) says we should run two commands to test if log2ram is really working.

This is my output, how do I know it is working as intended?

`root@MyCloud:~# df -h`
> Filesystem      Size  Used Avail Use% Mounted on
/dev/root       1.9G  566M  1.3G  32% /
devtmpfs        114M     0  114M   0% /dev
tmpfs           114M     0  114M   0% /dev/shm
tmpfs           114M  4.7M  109M   5% /run
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           114M     0  114M   0% /sys/fs/cgroup
tmpfs            64M     0   64M   0% /tmp
/dev/sda4       2.7T  1.5T  1.2T  56% /data
tmpfs            23M     0   23M   0% /run/user/0

`root@MyCloud:~# mount`
> /dev/md0 on / type ext3 (rw,noatime,nodiratime,errors=remount-ro,user_xattr,acl,commit=60,barrier=1,data=ordered)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,nosuid,nodev,noexec,relatime)
devtmpfs on /dev type devtmpfs (rw,nosuid,size=116056k,nr_inodes=29014,mode=755)
securityfs on /sys/kernel/security type securityfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620)
tmpfs on /run type tmpfs (rw,nosuid,nodev,mode=755)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,release_agent=/lib/systemd/systemd-cgroups-agent,name=systemd)
cgroup on /sys/fs/cgroup/cpuset type cgroup (rw,nosuid,nodev,noexec,relatime,cpuset)
cgroup on /sys/fs/cgroup/cpu,cpuacct type cgroup (rw,nosuid,nodev,noexec,relatime,cpuacct,cpu)
cgroup on /sys/fs/cgroup/memory type cgroup (rw,nosuid,nodev,noexec,relatime,memory)
cgroup on /sys/fs/cgroup/devices type cgroup (rw,nosuid,nodev,noexec,relatime,devices)
cgroup on /sys/fs/cgroup/net_cls type cgroup (rw,nosuid,nodev,noexec,relatime,net_cls)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,nosuid,nodev,noexec,relatime,blkio)
cgroup on /sys/fs/cgroup/perf_event type cgroup (rw,nosuid,nodev,noexec,relatime,perf_event)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
tmpfs on /tmp type tmpfs (rw,relatime,size=65536k)
/dev/sda4 on /data type ext4 (rw,noatime,nodiratime,errors=remount-ro,user_xattr,commit=60,barrier=0,data=writeback)
rpc_pipefs on /run/rpc_pipefs type rpc_pipefs (rw,relatime)
nfsd on /proc/fs/nfsd type nfsd (rw,relatime)
tmpfs on /run/user/0 type tmpfs (rw,nosuid,nodev,relatime,size=23228k,mode=700)

Sailing74, 2017-02-27:
I would guess it's not working, as I see no trace of the log2ram ramdisk. df-h and mount would typically show it. Are you sure systemd has started log2ram? What happens, if you start log2ram manually? There should be errors in your syslog/journal.

azlux, 2017-02-27:
You need the lines : 
for `df -h` : 
`log2ram                  40M  2,7M   38M   7% /var/log` depending if you have change some parameters

and for `mount` : 
`log2ram on /var/log type tmpfs (rw,nosuid,nodev,noexec,relatime,size=40960k,mode=755)`

azlux, 2017-02-27:
check `service log2ram status`is the service has start.
And watch the file `/var/log.hdd/log2ram.log` is there are errors !

hrvstr, 2017-02-27:
the log file is empty :/ 

the status shows the following:
```
root@MyCloud:~# service log2ram status
Showing one /org/freedesktop/systemd1/unit/log2ram_2eservice
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 object=/org/freedesktop/systemd1/unit/log2ram_2eservice interface=org.freedesktop.DBus.Properties member=GetAll cookie=1 reply_cookie=0 error=n/a
● log2ram.service - Log2Ram
   Loaded: loaded (/etc/systemd/system/log2ram.service; enabled)
   Active: failed (Result: exit-code) since Mon 2017-02-27 14:29:36 UTC; 4h 34min ago
  Process: 1121 ExecStart=/usr/local/bin/log2ram start (code=exited, status=203/EXEC)
 Main PID: 1121 (code=exited, status=203/EXEC)
Root directory /run/log/journal added.
Considering /run/log/journal/5f1d0e244c672cb2bffe767d5659627a.
Directory /run/log/journal/5f1d0e244c672cb2bffe767d5659627a added.
File /run/log/journal/5f1d0e244c672cb2bffe767d5659627a/system.journal added.
File /run/log/journal/5f1d0e244c672cb2bffe767d5659627a/system@6bc936f6b9e54404be290279730177fa-0000000000000ce9-00054985163f2c94.journal added.
File /run/log/journal/5f1d0e244c672cb2bffe767d5659627a/system@6bc936f6b9e54404be290279730177fa-0000000000000001-00054983e7b2fae4.journal added.
Journal filter: (((OBJECT_SYSTEMD_UNIT=log2ram.service AND _UID=0) OR (UNIT=log2ram.service AND _PID=1) OR (COREDUMP_UNIT=log2ram.service AND _UID=0 AND MESSAGE_ID=fc2e22bc6ee647b6b90729ab34a250b1) OR _SYSTEMD_UNIT=log2ram.service) AND _BOOT_ID=736a216b56314df09b07105f413a361b)

Feb 27 14:29:31 MyCloud systemd[1]: log2ram.service changed dead -> start
Feb 27 14:29:36 MyCloud systemd[1]: Child 1121 belongs to log2ram.service
Feb 27 14:29:36 MyCloud systemd[1]: log2ram.service: main process exited, c...EC
Feb 27 14:29:36 MyCloud systemd[1]: log2ram.service changed start -> failed
Feb 27 14:29:36 MyCloud systemd[1]: Job log2ram.service/start finished, res...ed
Feb 27 14:29:36 MyCloud systemd[1]: Failed to start Log2Ram.
Feb 27 14:29:36 MyCloud systemd[1]: Unit log2ram.service entered failed state.
Feb 27 14:29:36 MyCloud systemd[1]: log2ram.service: cgroup is empty
Root directory /run/log/journal removed.
Directory /run/log/journal/5f1d0e244c672cb2bffe767d5659627a removed.
mmap cache statistics: 1809 hit, 3 miss
Hint: Some lines were ellipsized, use -l to show in full.
```

when i try to start i get ...
```
root@MyCloud:~# service log2ram start 
Showing one /org/freedesktop/systemd1/unit/dbus_2esocket
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 object=/org/freedesktop/systemd1/unit/dbus_2esocket interface=org.freedesktop.DBus.Properties member=GetAll cookie=1 reply_cookie=0 error=n/a
Showing one /org/freedesktop/systemd1/unit/ssh_2esocket
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 object=/org/freedesktop/systemd1/unit/ssh_2esocket interface=org.freedesktop.DBus.Properties member=GetAll cookie=1 reply_cookie=0 error=n/a
Showing one /org/freedesktop/systemd1/unit/syslog_2esocket
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 object=/org/freedesktop/systemd1/unit/syslog_2esocket interface=org.freedesktop.DBus.Properties member=GetAll cookie=1 reply_cookie=0 error=n/a
Showing one /org/freedesktop/systemd1/unit/systemd_2dinitctl_2esocket
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 object=/org/freedesktop/systemd1/unit/systemd_2dinitctl_2esocket interface=org.freedesktop.DBus.Properties member=GetAll cookie=1 reply_cookie=0 error=n/a
Showing one /org/freedesktop/systemd1/unit/systemd_2djournald_2ddev_2dlog_2esocket
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 object=/org/freedesktop/systemd1/unit/systemd_2djournald_2ddev_2dlog_2esocket interface=org.freedesktop.DBus.Properties member=GetAll cookie=1 reply_cookie=0 error=n/a
Showing one /org/freedesktop/systemd1/unit/systemd_2djournald_2esocket
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 object=/org/freedesktop/systemd1/unit/systemd_2djournald_2esocket interface=org.freedesktop.DBus.Properties member=GetAll cookie=1 reply_cookie=0 error=n/a
Showing one /org/freedesktop/systemd1/unit/systemd_2dshutdownd_2esocket
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 object=/org/freedesktop/systemd1/unit/systemd_2dshutdownd_2esocket interface=org.freedesktop.DBus.Properties member=GetAll cookie=1 reply_cookie=0 error=n/a
Showing one /org/freedesktop/systemd1/unit/systemd_2dudevd_2dcontrol_2esocket
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 object=/org/freedesktop/systemd1/unit/systemd_2dudevd_2dcontrol_2esocket interface=org.freedesktop.DBus.Properties member=GetAll cookie=1 reply_cookie=0 error=n/a
Showing one /org/freedesktop/systemd1/unit/systemd_2dudevd_2dkernel_2esocket
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 object=/org/freedesktop/systemd1/unit/systemd_2dudevd_2dkernel_2esocket interface=org.freedesktop.DBus.Properties member=GetAll cookie=1 reply_cookie=0 error=n/a
Calling manager for StartUnit on log2ram.service, replace
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 object=/org/freedesktop/systemd1 interface=org.freedesktop.systemd1.Manager member=StartUnit cookie=1 reply_cookie=0 error=n/a
[... cut ...]
Got D-Bus request: org.freedesktop.DBus.Properties.PropertiesChanged() on /org/freedesktop/systemd1/job/626
Got message type=signal sender=n/a destination=n/a object=/org/freedesktop/systemd1 interface=org.freedesktop.systemd1.Manager member=JobRemoved cookie=8 reply_cookie=0 error=n/a
Got D-Bus request: org.freedesktop.systemd1.Manager.JobRemoved() on /org/freedesktop/systemd1
Job for log2ram.service failed. See 'systemctl status log2ram.service' and 'journalctl -xn' for details.
Got result failed/Input/output error for job log2ram.service
```

azlux, 2017-02-27:
Wow. It's fail but don't show us the real error.
what are the output of `systemctl status log2ram.service`

What is the output of `sudo /usr/local/bin/log2ram start` ?

hrvstr, 2017-02-27:
```
root@MyCloud:~# systemctl status log2ram.service
Showing one /org/freedesktop/systemd1/unit/log2ram_2eservice
Sent message type=method_call sender=n/a destination=org.freedesktop.systemd1 object=/org/freedesktop/systemd1/unit/log2ram_2eservice interface=org.freedesktop.DBus.Properties member=GetAll cookie=1 reply_cookie=0 error=n/a
● log2ram.service - Log2Ram
   Loaded: loaded (/etc/systemd/system/log2ram.service; enabled)
   Active: failed (Result: exit-code) since Mo 2017-02-27 19:05:20 UTC; 1h 34min ago
  Process: 4161 ExecStart=/usr/local/bin/log2ram start (code=exited, status=203/EXEC)
 Main PID: 4161 (code=exited, status=203/EXEC)
Root directory /run/log/journal added.
Considering /run/log/journal/5f1d0e244c672cb2bffe767d5659627a.
Directory /run/log/journal/5f1d0e244c672cb2bffe767d5659627a added.
File /run/log/journal/5f1d0e244c672cb2bffe767d5659627a/system.journal added.
File /run/log/journal/5f1d0e244c672cb2bffe767d5659627a/system@6bc936f6b9e54404be290279730177fa-0000000000000ce9-00054985163f2c94.journal added.
File /run/log/journal/5f1d0e244c672cb2bffe767d5659627a/system@6bc936f6b9e54404be290279730177fa-0000000000000001-00054983e7b2fae4.journal added.
Journal filter: (((OBJECT_SYSTEMD_UNIT=log2ram.service AND _UID=0) OR (UNIT=log2ram.service AND _PID=1) OR (COREDUMP_UNIT=log2ram.service AND _UID=0 AND MESSAGE_ID=fc2e22bc6ee647b6b90729ab34a250b1) OR _SYSTEMD_UNIT=log2ram.service) AND _BOOT_ID=736a216b56314df09b07105f413a361b)

Feb 27 19:05:20 MyCloud systemd[1]: Forked /usr/local/bin/log2ram as 4161
Feb 27 19:05:20 MyCloud systemd[1]: log2ram.service changed failed -> start
Feb 27 19:05:20 MyCloud systemd[1]: Child 4161 belongs to log2ram.service
Feb 27 19:05:20 MyCloud systemd[1]: log2ram.service: main process exited, c...EC
Feb 27 19:05:20 MyCloud systemd[1]: log2ram.service changed start -> failed
Feb 27 19:05:21 MyCloud systemd[1]: Job log2ram.service/start finished, res...ed
Feb 27 19:05:21 MyCloud systemd[1]: Failed to start Log2Ram.
Feb 27 19:05:21 MyCloud systemd[1]: Unit log2ram.service entered failed state.
Feb 27 19:05:21 MyCloud systemd[1]: log2ram.service: cgroup is empty
Root directory /run/log/journal removed.
Directory /run/log/journal/5f1d0e244c672cb2bffe767d5659627a removed.
mmap cache statistics: 3847 hit, 3 miss
Hint: Some lines were ellipsized, use -l to show in full.
```

```
root@MyCloud:~# /usr/local/bin/log2ram start
-bash: /usr/local/bin/log2ram: Ist ein Verzeichnis (is a directory)
```

azlux, 2017-02-27:
Read the full output please 'Hint: Some lines were ellipsized, use -l to show in full.'
/usr/local/bin/log2ram shoudn't be a directory, it's where the log2ram script must be

hrvstr, 2017-02-27:
ah okay that was the issue. I somehow copied the whole folder from github there. when I placed only the log2ram file in /usr/local/bin/ it started successfully.

closed 2017-02-27 by hrvstr

azlux, 2017-02-27:
there are a install script. use it.