azlux/log2ram #13: Write to HDD. Open question !

azlux, 2017-04-13:
Hi, 
I am wondering, what is the writing frequency on the hard disk of the `/var/log` folder a normal user really needs? (average user like small server or desktop usage)

I've heard sometime 2hours, and sometime 1day.

@zador-blood-stained @NickDaly

zador-blood-stained, 2017-04-13:
In my opinion it depends on how much data in `/var/log` a user is willing to loose in case of a power loss.

It is possible to use a configuration file `/etc/default/log2ram` and add an option to select at least between 2 variants - hourly and daily sync, by installing scripts both to `/etc/cron.hourly` and `/etc/cron.daily` and changing them like this:
```
#! /bin/sh

. /etc/default/log2ram

# for cron.daily
[ "$DAILY_SYNC" = true ] && /usr/local/bin/log2ram write
# for cron.hourly
[ "$DAILY_SYNC" != true ] && /usr/local/bin/log2ram write
```

NickDaly, 2017-04-27:
I just use hourly sync on my device.  It saves my laptop's old harddrive from spinning down and up again every other minute.

Of course, on my other computer that has Gnome's Tracker (desktop indexer) enabled, the disk never spins down because it's always reindexing something.

closed 2017-05-13 by azlux