azlux/log2ram #15: Few question about log2ram if it is full

easyyu, 2017-05-13:
I hadn't idea where else to ask, so I will ask here.
I understand the logic of log2ram and the idea is super.
But, I can not realize what will happen when the partitin of (for example of 40MB) will be filled!?
I have some app what doing heavy loging and it fill in few hours whole log partition.
I understand that that log partiton is written to SD/HDD with crond every hours, but my loging stop when partiton is full.
OK, the solution is to edit the conf file and reserve more MB for it, but is there any solution that if it reach for example 90% to write it to SD/hdd log and **CLEAR/FORMAT** log2ram partiton to prevent stop of loging?

Thx for suggestion!

azlux, 2017-05-13:
This is the same question as "What happens to the logs if my hard drive is full". The syslog simply can not write anymore(But log2ram can always copy the logs to the hard disk without any problem). The worries do not come from log2ram, but from the management of a full disk (here the one log2ram mount).
We cannot change the size of a mounted partition without unmount it. Application will try to write and fail during the process, and If you have unmount it, you will need to restart because application will have into memory the old inode number and will still try to write log into this file. It will be the same problem if I clean it, some application will not be able to write instead you restart it).
So the program who will manage that is your **logrotate** (you can specify a max size for a log file) because logrotate can make rotation without changing the inode number.

Hope it's help.

easyyu, 2017-05-13:
First of all thax for clarification, and for replay.
I can not tell that it is same couse in this case you are limited with the size of HW RAM, but in case with hdd you can always have a lot of space, but the logic of what will happen when partiton become full is same.
Yes I'm using logrotate, and have personal modification of roatting files.

I think that I will write some script and put in crontab, what checking the size of log2ram and in case that it become bigger than x% to delete oldest files (gz, then log).

Sailing74, 2017-05-13:
Deleting .log won't necessarily help. Often, the logs are kept open, so the app continues to write to file, even if you think you already deleted it (or deleting doesn't work in the first place).

Buy more RAM, reduce logging, or log to disk, if you need to play safer.

easyyu, 2017-05-14:
Thx..understand..
Logrotate is the best options then.

closed 2017-05-14 by azlux