azlux/log2ram #16: Does it work in init.d sytems?

heikoh81, 2017-05-13:
Hi,

thanks for this great tool.
Currently, I use ramlog on Rasbian Wheezy systems with init.d

Is there a way I could use log2ram on init.d-systems as well?
Maybe you could provdie an init-script for older systems.
(ramlog has not been updated for years, and sometimes making trouble on my raspberrys...)

Greetings,
Heiko (Germany)

azlux, 2017-05-13:
Fast answer : yes. The functions will work because I use natives tools bit you need to modify 2 file. Long answer this evening when I will have my computer.

azlux, 2017-05-13:
So the only file you need to change is the [service file](https://github.com/azlux/log2ram/blob/master/log2ram.service). You need to transform it into an init.d file. Try to follow the same idea (start before the syslog, and write before the reboot). After that you can modify the install file to install the new file into the /etc/init.d folder. 
If you are motivated to make the init.d file, fork the project and I will be able to manager both systemd and init.d system

closed 2017-09-12 by azlux