azlux/log2ram #17: after install apache2 cant start

mardzo, 2017-05-30:
It seems apache2 needs /var/log/apache2/. log2ram transfers this directory to /var/log.hdd/apache2. 

I fixed it by adding :
@reboot sudo mkdir /var/log/apache2
@reboot sudo service apache2 restart
in crontab but there must be a better way?

How to fix this so apache2 can find its log folder on reboot?

azlux, 2017-05-30:
Hi. Do you use the last version here ? Log2ram copy back the log into ram at startup usually. 
Please check you /var/log/message.log to check if there are any error from log2ram at startup.
Can you give me 'uname -a' info ?

mardzo, 2017-05-30:
Linux raspberrypi 4.4.15-v7+ #898 SMP Thu Jul 21 14:54:14 BST 2016 armv7l GNU/Linux

azlux, 2017-05-30:
Log2ram start before because it start before the rsyslog system.
You can try to add apache2.service next to other servive on the Before parameter  into  /etc/systemd/system/log2ram.service it will solve the pb

Can you copy/paste your /var/log/message here : https://azlux.fr/bin/ and send the link?

mardzo, 2017-05-30:
I have made this change:
Before=rsyslog.service apache2.service syslog.target systemd-journald.service sysinit.target shutdown.target
It still does now work.  About the log i get this: cannot open ‘/var/log/message’ for reading: No such file or directory

azlux, 2017-05-30:
No trace here.
Can i have the output of  `systemd-analyze plot `
And `systemd-analyze critical-chain` i suspect  apache starting before rsyslog.
And please use  [a past bin](https://azlux.fr/bin) for the first output, it's a csv picture.

azlux, 2017-06-06:
Is you /var/log.hdd/apache2 folder is into this path ?

azlux, 2017-06-14:
Can you check If you still have this issue (not problem on my VM)

closed 2017-07-17 by azlux