azlux/log2ram #18: Mosquitto and Apache

scargill, 2017-06-11:
I'm only just realised that log2ram is causing issues .... so despite using a recent version I discovered at some point that Mosquitto service was not starting up properly - I could run Mosquitto normally - but not as the service....  then in the process of that I discovered that the Apache log folder had disappeared and of course without that Apache won't start.

I recreated the Apache log folder, took out log2ram, rebooted and now everything is working a fine  - I really do need log2ram to work to reduce writes to SD - but it is causing these issues - or probably more correctly the interaction between Apache and log2ram - and Mosquitto and log2ram - is causing problems.

azlux, 2017-06-11:
Can i have the output of  `systemd-analyze plot `
And `systemd-analyze critical-chain` i suspect  apache starting before rsyslog.
And please use  [a past bin](https://azlux.fr/bin) for the first output, it's a svg picture.   
You can also check the /var/log.hdd/log2ram.log file if there are any error here.

scargill, 2017-06-11:
There’s a lot

 

systemd-analyze plot

<?xml version="1.0" standalone="no"?>

<!DOCTYPE svg PUBLIC "-//W3C//DTD SVG 1.1//EN" "http://www.w3.org/Graphics/SVG/1.1/DTD/svg11.dtd">

<svg width="3147px" height="3310px" version="1.1" xmlns="http://www.w3.org/2000/svg">

 

 

 

<defs>

  <style type="text/css">

    <![CDATA[

      rect       { stroke-width: 1; stroke-opacity: 0; }

      rect.background   { fill: rgb(255,255,255); }

      rect.activating   { fill: rgb(255,0,0); fill-opacity: 0.7; }

      rect.active       { fill: rgb(200,150,150); fill-opacity: 0.7; }

      rect.deactivating { fill: rgb(150,100,100); fill-opacity: 0.7; }

      rect.kernel       { fill: rgb(150,150,150); fill-opacity: 0.7; }

      rect.initrd       { fill: rgb(150,150,150); fill-opacity: 0.7; }

      rect.firmware     { fill: rgb(150,150,150); fill-opacity: 0.7; }

      rect.loader       { fill: rgb(150,150,150); fill-opacity: 0.7; }

      rect.userspace    { fill: rgb(150,150,150); fill-opacity: 0.7; }

      rect.security     { fill: rgb(144,238,144); fill-opacity: 0.7; }

      rect.generators   { fill: rgb(102,204,255); fill-opacity: 0.7; }

      rect.unitsload    { fill: rgb( 82,184,255); fill-opacity: 0.7; }

      rect.box   { fill: rgb(240,240,240); stroke: rgb(192,192,192); }

      line       { stroke: rgb(64,64,64); stroke-width: 1; }

//    line.sec1  { }

      line.sec5  { stroke-width: 2; }

      line.sec01 { stroke: rgb(224,224,224); stroke-width: 1; }

      text       { font-family: Verdana, Helvetica; font-size: 14px; }

      text.left  { font-family: Verdana, Helvetica; font-size: 14px; text-anchor: start; }

      text.right { font-family: Verdana, Helvetica; font-size: 14px; text-anchor: end; }

      text.sec   { font-size: 10px; }

    ]]>

[... 1554 lines cut ...]
  <rect class="deactivating" x="3067.402" y="2160.000" width="0.000" height="19.000" />

  <text class="left" x="793.605" y="2174.000">systemd-logind.service (381ms)</text>

  <rect class="activating" x="789.568" y="2180.000" width="16.594" height="19.000" />

  <rect class="active" x="806.162" y="2180.000" width="2261.240" height="19.000" />

  <rect class="deactivating" x="3067.402" y="2180.000" width="0.000" height="19.000" />

  <text class="left" x="794.568" y="2194.000">mosquitto.service (165ms)</text>

  <rect class="activating" x="790.967" y="2200.000" width="302.369" height="19.000" />

  <rect class="active" x="1093.336" y="2200.000" width="1974.067" height="19.000" />

  <rect class="deactivating" x="3067.402" y="2200.000" width="0.000" height="19.000" />

  <text class="left" x="795.967" y="2214.000">dphys-swapfile.service (3.023s)</text>

  <rect class="activating" x="792.658" y="2220.000" width="32.091" height="19.000" />

  <rect class="active" x="824.749" y="2220.000" width="2242.654" height="19.000" />

  <rect class="deactivating" x="3067.402" y="2220.000" width="0.000" heigh

scargill, 2017-06-11:
Here it is – sorry – as requested.

 

 <https://azlux.fr/bin/?bb2ecea07e3000df#b6/ZO1xeHpu7YK5IxoRqbC3k4/tRKNORAInWd5Ffn5s=> https://azlux.fr/bin/?bb2ecea07e3000df#b6/ZO1xeHpu7YK5IxoRqbC3k4/tRKNORAInWd5Ffn5s=

 

There is no /var/log.hdd/log2ram but I disabled it as the MQTT is running my house :)

 

So what next…

azlux, 2017-06-11:
Hummm... nothing here, 
You can have a better output with systemd-analyze blame to see the start sequence yourself.
I try a fix with 9d76a1b08219ce5b4492dedf9eedc9851060705e Can you try to reinstall it (with the install script) ?
PS : it's hard to diagnose without the log2ram service enable.

azlux, 2017-06-14:
Can you check If you still have this issue (not problem on my VM)

closed 2017-10-02 by azlux