azlux/log2ram #20: write() should check tmpfs free space

Sailing74, 2017-07-14:
I just noticed that log2ram had run out of space here, because my logs ballooned following massive spammer attacks onto my server.

As one consequence, I just suggested a change to fix the situation where, upon startup, it's already clear that tmpfs will run out of space. In that case, not only should log2ram exit, but the binds should be removed. Otherwise, we'll have no synched log data on tmpfs, but new logs will still be written there, resulting in loss of historical logs.

Now, to be safer, log2ram should also check load status of tmpfs regularly, probably during write(). And then, if an issue is discovered, ideally log2ram would exit, but not sure if that's feasible with probably lots of file writes open...? Alternatively, log2ram should at least issue some noticeable warnings in such situation. 

Will think about this some more when I find the time... Or maybe someone else has some bright idea...

azlux, 2017-07-17:
I think that is the issue for #19

azlux, 2017-07-17:
Fixed with #19 and https://github.com/azlux/log2ram/commit/2e55fa7bed4774c9e08b60c7f1bd1f98cec3301d

closed 2017-07-17 by azlux