azlux/log2ram #21: Uninstall script

ymettier, 2017-07-23:
Hello,

Do you have an uninstall.sh script ? (*)

There are 2 ideas for this question :
1/ I'm working with Ansible (not install.sh). The idea is to write the playbook and test it, uninstall, re-test it and so on.

2/ maybe it could become a debian package for Raspbian ?

(*) yes, such a question here sounds weired :)

azlux, 2017-07-23:
Hi @ymettier ,
Yes, it's on my DoTo list.
The uninstall file is easy to do, but I just need to take time for it.
(I will do it ASAP now someone ask for it)
I have a Debian package, but no tested, and not really updated (into realease), but I will stop to create the package, I don't know enough about packaging to do it myself.

Az

azlux, 2017-07-27:
Uninstall script done (but no tested).
I've never work with Ansible. I don't know was you need to deploy on it. Open issues if you need help/feature/improvement.

closed 2017-08-07 by azlux