azlux/log2ram #24: cron emails daily

lampra1, 2017-08-21:
Hi, I use armbian on cubietruck(a20). Since the the last update log2ram complained for folder /var/log.hdd/ which I created. Since then, I receive daily the email below from root.
In order not to receive this email, as a workaround, I included > /dev/null 2>&1 in the crontab for the daily job. Nevertheless, this workaround affects all daily jobs. 
What is the correct way  to stop receiving daily emails  from log2ram if there is no error?
Thank you
```
/etc/cron.daily/log2ram:
sending incremental file list
./
alternatives.log
auth.log
daemon.log
dpkg.log
kern.log
lastlog
mail.info
mail.log
messages
minidlna.log
mysql.log
mysql.log.1.gz
mysql.log.2.gz
mysql.log.3.gz
mysql.log.4.gz
mysql.log.5.gz
mysql.log.6.gz
mysql.log.7.gz
syslog
syslog.1
syslog.2.gz
syslog.3.gz
syslog.4.gz
syslog.5.gz
syslog.6.gz
syslog.7.gz
wtmp
apache2/
apache2/access.log
apache2/access.log.1
apache2/access.log.10.gz
apache2/access.log.11.gz
apache2/access.log.12.gz
apache2/access.log.13.gz
apache2/access.log.14.gz
apache2/access.log.2.gz
apache2/access.log.3.gz
apache2/access.log.4.gz
apache2/access.log.5.gz
apache2/access.log.6.gz
apache2/access.log.7.gz
apache2/access.log.8.gz
apache2/access.log.9.gz
apache2/error.log
apache2/error.log.1
apache2/error.log.10.gz
apache2/error.log.11.gz
apache2/error.log.12.gz
apache2/error.log.13.gz
apache2/error.log.14.gz
apache2/error.log.2.gz
apache2/error.log.3.gz
apache2/error.log.4.gz
apache2/error.log.5.gz
apache2/error.log.6.gz
apache2/error.log.7.gz
apache2/error.log.8.gz
apache2/error.log.9.gz
apt/history.log
apt/term.log
mysql/
mysql/error.log
mysql/error.log.1.gz
mysql/error.log.2.gz
mysql/error.log.3.gz
mysql/error.log.4.gz
mysql/error.log.5.gz
mysql/error.log.6.gz
mysql/error.log.7.gz
unattended-upgrades/unattended-upgrades-dpkg.log
unattended-upgrades/unattended-upgrades.log

sent 5,106,397 bytes  received 1,360 bytes  3,405,171.33 bytes/sec
total size is 9,042,498  speedup is 1.77`
```

azlux, 2017-08-22:
Hi.
If we put you command instead of the actual one. You will have no mail if error occurs with log2ram. The script shouldn't send mail if no error.
Can you chek check /var/log.hdd/log2ram.log ? And if you have same write access on /var/log and /var/log.hdd ?

MegaV0lt, 2017-08-22:
The script mails every output even non errors. to make rsync only output error-messages you must add the quiet-switch:

` -q, --quiet                 suppress non-error messages
`

lampra1, 2017-08-22:
Well, I think that  MegaV0lt is correct.  According to cron(8) manual "When executing commands, any output is mailed to the owner of the crontab". To my knowledge, rsync produces output with or without errors. 
I changed the script and added the the quiet-switch.
I will report back if any problem occurs.

MegaV0lt, 2017-08-22:
In my other project i use rsync for backup. i redirect stdout to /dev/null and errors to a logfile. So no output on screen or cron... See --log-file for normal output:
`rsync "${RSYNC_OPT[@]}" --log-file="$LOG" --exclude-from="$EXFROM" \
            --backup-dir="$BAK_DIR" "${MAN_SOURCE[@]}" "$R_TARGET" >/dev/null 2>> "$ERRLOG"
`
For reference: https://github.com/MegaV0lt/MV_Backup

azlux, 2017-08-27:
For cron you don't need to pipe through mail. The cron daemon will automatically mail any output of your command to you. Your crontab entry should look like : [here my case](https://github.com/azlux/log2ram/blob/master/log2ram.hourly):
```
 ./prog >/dev/null
```
If there is no STDERR output, you won't get any mail. So if you have mail, it's because the output come from STDERR and not STDOUT.
the `-q` will avaoid rsync to send STDOUT if no error, but that is the goal of `>/dev/null` so I still don't know why you have this error.
@MegaV0lt you push your stderr into the $ERRLOG file, but I don't need that, the email from cron is good.

MegaV0lt, 2017-08-28:
You are right. In my example the logs get collected and mailed later... I think lampra1 doesn't have the >/dev/null in the cron

lampra1, 2017-09-12:
Today, I saw this topic on armbian [https://forum.armbian.com/index.php?/topic/3728-varlog-varloghdd/](url).
I checked and log2ram.service was disabled on my system. I enabled again the service and removed the `-q` (quiet) switch in the script. I will report back here if the daily email problem occurs again.  
That might be the root of the problem, sending emails everyday which is not a issue of the script itself but an upgrade handling issue. It would be great if the script could produce some other more informative output in such a case.

azlux, 2017-09-12:
@lampra1 Can you open a new issue for that ? it's maybe not the same issue

closed 2017-09-12 by azlux