azlux/log2ram #29: cron witx exti code

MegaV0lt, 2017-10-09:
since last update i get a mail from cron with this content:

> run-parts: /etc/cron.daily/log2ram exited with return code 1

azlux, 2017-10-09:
The only place where log2ram do an `exit 1` is when it check is : 
```
[ -d $HDD_LOG/ ] || exit 1
```
whith `$HDD_LOG = /var/hdd.log`
May be you can do : `  [ -d /var/hdd.log/ ] || echo "test" `to check if the condition fail or not.

Can you check the log2ram log file `/var/hdd.log/log2ram.log` ?

azlux, 2017-10-09:
The other command producing the `error 1` is the free space free.
```
if [ ! -z `du -sh -t 40M /var/hdd.log/ | cut -f1` ];then
   echo "ERROR: RAM disk too small. Can't sync."
   exit 1
fi
```
Have you an echo when you copy/paste theses 4 lines ?

MegaV0lt, 2017-10-09:
Uh...
`root auf HP-T610 am 09.10.2017 11:51 
[~] # if [ ! -z `du -sh -t 40M /var/hdd.log/ | cut -f1` ];then
>    echo "ERROR: RAM disk too small. Can't sync."
>    exit 1
> fi
du: Zugriff auf '/var/hdd.log/' nicht möglich: Datei oder Verzeichnis nicht gefunden
`
is not matching the actual dirs:
`[~] # ll /var
insgesamt 40K
drwxr-xr-x  2 root     root     4,0K Okt  9 00:05 backups
drwxr-xr-x 17 root     root     4,0K Okt  8 10:09 cache
drwxr-xr-x 55 root     root     4,0K Okt  8 10:09 lib
drwxrwsr-x  2 root     root     4,0K Nov 30  2014 local
lrwxrwxrwx  1 root     root        9 Aug 13 22:38 lock -> /run/lock
drwxr-xr-x 13 root     root     2,9K Okt  9 05:19 log
drwxr-xr-x 13 root     root     4,0K Okt  3 23:02 log.hdd
drwxrwsr-x  2 root     root     4,0K Apr 25  2015 mail
drwxr-xr-x  2 root     root     4,0K Apr 25  2015 opt
lrwxrwxrwx  1 root     root        4 Aug 13 22:38 run -> /run
drwxr-xr-x  7 root     root     4,0K Mai  3  2015 spool
drwxrwxrwt  4 root     root     4,0K Okt  9 11:39 tmp
drwxr-xr-x  7 www-data www-data 4,0K Okt  9 09:31 www
`

in install.sh it makes the dir /var/log.hdd but the log2ram works wit /var/hdd.log !

So pleas make this uniqu by changing the dir in log2ram to teh /var/log.hdd

MegaV0lt, 2017-10-09:
After reboot looks fine:
`root auf HP-T610 am 09.10.2017 12:10 
[~] # ll /var
insgesamt 40K
drwxr-xr-x  2 root     root     4,0K Okt  9 00:05 backups
drwxr-xr-x 17 root     root     4,0K Okt  8 10:09 cache
drwxr-xr-x 13 root     root     4,0K Okt  3 23:02 hdd.log
drwxr-xr-x 55 root     root     4,0K Okt  8 10:09 lib
drwxrwsr-x  2 root     root     4,0K Nov 30  2014 local
lrwxrwxrwx  1 root     root        9 Aug 13 22:38 lock -> /run/lock
drwxr-xr-x 13 root     root     2,9K Okt  3 23:02 log
drwxrwsr-x  2 root     root     4,0K Apr 25  2015 mail
drwxr-xr-x  2 root     root     4,0K Apr 25  2015 opt
lrwxrwxrwx  1 root     root        4 Aug 13 22:38 run -> /run
drwxr-xr-x  7 root     root     4,0K Mai  3  2015 spool
drwxrwxrwt  4 root     root     4,0K Okt  9 12:09 tmp
drwxr-xr-x  7 www-data www-data 4,0K Okt  9 09:31 www
root auf HP-T610 am 09.10.2017 12:10 
[~] # log2ram write
sending incremental file list
OSEmu.log
auth.log
daemon.log
debug
fail2ban.log
kern.log
lastlog
mail.err
mail.info
mail.log
mail.warn
messages
monitorix
oscam.log
oscam_status.log
sendmail.log
syslog
user.log
wtmp
apache2/access.log
apache2/error.log
apache2/other_vhosts_access.log
mysql/error.log
samba/log.nmbd
samba/log.smbd

sent 2,167,505 bytes  received 514 bytes  4,336,038.00 bytes/sec
total size is 24,452,049  speedup is 11.28
`

azlux, 2017-10-09:
You didn't run the `install.sh` file when you have upgrade.
The install have this code : 
```
  if [ -d /var/log.hdd ]; then
    rm -r /var/log.hdd
fi
```
This remove the old folder you have. And create the new folder for the mount point.

closed 2017-10-19 by azlux