azlux/log2ram #35: ERROR: RAM disk too small. Can't sync. ?

disquette38, 2017-11-13:
Hello, 
I am trying to install log2ram no error messages in the installation, but unable to run log2ram "ERROR: RAM disk too small." Can not sync. " here is the result of df -h

/dev/root          7.1G    1.7G  5.2G  25% /
devtmpfs           458M       0  458M   0% /dev
tmpfs              462M    4.0K  462M   1% /dev/shm
tmpfs              462M    6.5M  456M   2% /run
tmpfs              5.0M    4.0K  5.0M   1% /run/lock
tmpfs              462M       0  462M   0% /sys/fs/cgroup
/dev/mmcblk0p1      63M     21M   42M  34% /boot

 How to fix this problem ?

azlux, 2017-11-13:
Hi,
Check the SIZE variable into the configuration file (/etc/log2ram.conf)
and check the /var/log folder size : `du -sh /var/log`
If you try to put more data than the size reserved into the RAM, you get this error.

disquette38, 2017-11-13:
👍 yesssss it's good :) du -sh /var/log -> **103M** 
The SIZE variable into the configuration file (/etc/log2ram.conf) **150M** 
thank !

azlux, 2017-11-13:
103Mo of log ? wow !
Don't forget to user logrotate !

closed 2017-11-13 by azlux