azlux/log2ram #38: systemctl --failed : console-setup

ymettier, 2017-12-16:
Hello,

I'm now using log2ram for several weeks on my raspberry pi.
I noticed that when I restart my raspberry, I have a failure :
`systemctl --failed` shows `console-setup.service`.

I suggest that you add `console-setup.service` in `log2ram.service` in the line `Before=`.
I did it : it works for me.

Regards,
Yves

azlux, 2017-12-16:
Can i have the output of `systemd-analyze plot`
And `systemd-analyze critical-chain`
And please use a paste bin for the first output, it's a svg picture. ([this one for example](https://azlux.fr/bin)) and can be really big.
I want to understand why `console-setup`block the mount.
Do you have any error into `/var/hdd.log/log2ram.log`

ymettier, 2017-12-17:
```
# systemctl --failed
  UNIT                  LOAD   ACTIVE SUB    DESCRIPTION                
● console-setup.service loaded failed failed Set console font and keymap

LOAD   = Reflects whether the unit definition was properly loaded.
ACTIVE = The high-level unit activation state, i.e. generalization of SUB.
SUB    = The low-level unit activation state, values depend on unit type.

1 loaded units listed. Pass --all to see loaded but inactive units, too.
To show all installed unit files use 'systemctl list-unit-files'.
```
`systemd-analyze plot` : 
https://azlux.fr/bin/?9cd7d648ca961c01#xT0INUlM7veVNwJqzTKymhMtlv57CAemU1j+2Rvenpg=

log2ram starts before console-setup, but seems to finish after the start of console-setup.

```
# systemd-analyze critical-chain
The time after the unit is active or started is printed after the "@" character.
The time the unit takes to start is printed after the "+" character.

graphical.target @18.409s
└─multi-user.target @18.409s
  └─docker.service @14.605s +3.801s
    └─network-online.target @14.593s
      └─network.target @14.587s
        └─dhcpcd.service @3.888s +10.696s
          └─basic.target @3.858s
            └─sockets.target @3.857s
              └─docker.socket @3.840s +16ms
                └─sysinit.target @3.839s
                  └─systemd-timesyncd.service @3.483s +350ms
                    └─systemd-tmpfiles-setup.service @3.396s +78ms
                      └─systemd-journal-flush.service @3.322s +64ms
                        └─systemd-journald.service @3.146s +155ms
                          └─log2ram.service @3.002s +138ms
                            └─var-hdd.log.mount @3.074s
                              └─local-fs-pre.target @2.718s
                                └─lvm2-monitor.service @1.067s +1.650s
                                  └─lvm2-lvmetad.service @1.304s
                                    └─systemd-journald.socket @1.015s
                                      └─-.mount @808ms
                                        └─system.slice @927ms
                                          └─-.slice @808ms
```

```
# stat /var/hdd.log/log2ram.log
  File: /var/hdd.log/log2ram.log
  Size: 0         	Blocks: 0          IO Block: 4096   regular empty file
[...]
```
The log file is empty.

```
# lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.1 (stretch)
Release:	9.1
Codename:	stretch
```

I hope this helps.

ymettier, 2017-12-17:
Bad news...

WIth console-setup.service in the `Before=` line I have the same problem.

I reboot my raspberry pi :

```
# cat /etc/systemd/system/log2ram.service 
[Unit]
Description=Log2Ram
DefaultDependencies=no
Before=basic.target rsyslog.service syslog.target systemd-journald.service sysinit.target shutdown.target console-setup.service apache2.service
After=local-fs.target
Conflicts=shutdown.target reboot.target halt.target
RequiresMountsFor=/var/log /var/hdd.log
IgnoreOnIsolate=yes

[Service]
Type=oneshot
ExecStart= /usr/local/bin/log2ram start
ExecStop= /usr/local/bin/log2ram stop
ExecReload= /usr/local/bin/log2ram write
RemainAfterExit=yes

[Install]
WantedBy=sysinit.target
root@pifour:/home/yves# systemctl -failed
systemctl: invalid option -- 'e'
root@pifour:/home/yves# cat /etc/systemd/system/log2ram.service 
[Unit]
Description=Log2Ram
DefaultDependencies=no
Before=basic.target rsyslog.service syslog.target systemd-journald.service sysinit.target shutdown.target console-setup.service apache2.service
After=local-fs.target
Conflicts=shutdown.target reboot.target halt.target
RequiresMountsFor=/var/log /var/hdd.log
IgnoreOnIsolate=yes

[Service]
Type=oneshot
ExecStart= /usr/local/bin/log2ram start
ExecStop= /usr/local/bin/log2ram stop
ExecReload= /usr/local/bin/log2ram write
RemainAfterExit=yes

[Install]
WantedBy=sysinit.target
```
```
# systemctl --failed
  UNIT                  LOAD   ACTIVE SUB    DESCRIPTION                
● console-setup.service loaded failed failed Set console font and keymap

LOAD   = Reflects whether the unit definition was properly loaded.
ACTIVE = The high-level unit activation state, i.e. generalization of SUB.
SUB    = The low-level unit activation state, values depend on unit type.

1 loaded units listed. Pass --all to see loaded but inactive units, too.
To show all installed unit files use 'systemctl list-unit-files'.
```
```
# systemctl restart console-setup.service
```
```
# systemctl --failed
0 loaded units listed. Pass --all to see loaded but inactive units, too.
To show all installed unit files use 'systemctl list-unit-files'.
```
```
# systemd-analyze critical-chain |cat
The time after the unit is active or started is printed after the "@" character.
The time the unit takes to start is printed after the "+" character.

graphical.target @19.923s
└─multi-user.target @19.923s
  └─docker.service @16.084s +3.838s
    └─network-online.target @16.077s
      └─network.target @16.046s
        └─dhcpcd.service @3.963s +12.081s
          └─basic.target @3.781s
            └─sockets.target @3.779s
              └─docker.socket @3.748s +28ms
                └─sysinit.target @3.746s
                  └─systemd-timesyncd.service @3.407s +338ms
                    └─systemd-tmpfiles-setup.service @3.295s +91ms
                      └─systemd-journal-flush.service @3.229s +59ms
                        └─systemd-journald.service @3.085s +138ms
                          └─log2ram.service @2.944s +135ms
                            └─var-hdd.log.mount @2.992s
                              └─local-fs-pre.target @2.677s
                                └─lvm2-monitor.service @1.052s +1.624s
                                  └─lvm2-lvmetad.service @1.275s
                                    └─systemd-journald.socket @992ms
                                      └─-.mount @808ms
                                        └─system.slice @906ms
                                          └─-.slice @808ms
```

https://azlux.fr/bin/?a7dae9364be4175e#zNaJq0oplFxC0MXg/vU74CFtbAVig7XBZw0P3LhhWBY=

Can't find console-setup on this one...

log file is empty.

So adding console-setup in `Before=` line is not the solution.

Do you have a better fix ?

azlux, 2017-12-17:
Be carefull, all you output from `systemd-analyze plot` are broken, you don't have all the data. I cannot create the svg with that :(
use `systemd-analyze plot > out.sgv` to redirect all stdout into a file.

azlux, 2018-02-07:
another question have you the log of console-setup.service ?
`journalctl -u console-setup.service -b`

azlux, 2018-04-30:
I will close this issue If I cannot have answer to debug it !

ymettier, 2018-05-01:
Oops, I'm sorry, I had to put my raspberry in a "production" mode and I could not play with it to debug.
I still have the problem, but I have a "restart_services.sh" script (with 4 or 5 lines of `systemctl restart <service>` for each of my services) : I run it each time I restart my raspberry pi. I wrote that script because I had no time to debug myself and I have the feeling that I have to understand what happens on my raspberry pi and why it does not happen on yours or on any other raspberry.

I understand that you do not want to keep this issue open with no feedback. Maybe the best is to close it with a message "cannot reproduce and not enough feedback". And when I have better feedback, I reopen it (I have a new raspberry pi; I just need some time to play with it again) :)

azlux, 2018-05-01:
I will close this ticket. Feel free to reopen it again if you can reproduce the issue.
you can add all your services at this line of the log2ram.service `Before=`, this can help.

closed 2018-05-01 by azlux