azlux/log2ram #45: Current rsync usage may cause excessive SD card writes

tim-seoss, 2018-01-14:
The way the script uses rsync currently, results in the following (I think):

If log file has been modified since the last rsync, then copy the file on the SD card to a temporary file on the SD card, append to this temporary file, remove the original log file on the SD card, and finally rename the temporary file to the correct filename.

Invoking rsync with the `--inplace` arguement would result in:

If the log file has been modified since the last rsync, append the contents to the existing file on the SD card.

My guess is that using `--inplace` is likely to cause less SD card wear, BICBW.  It might be a good idea to use the /sys filesystem I/O counters to check the total write IOPS with each use case...

azlux, 2018-01-14:
Hi,
Thank for the information, I will check this parameter of rsync.
I'm not sure of what's append when a compress log will change from log.1.gz to log.2.gz with this option.

tim-seoss, 2018-01-15:
Probably rotated files (e.g. log.1.gz -> log.2.gz) will be deleted and recreated with both versions, but  `--fuzzy --delete-after` might help with rotated files?

tim-seoss, 2018-01-15:
Maybe two-stage is best - doing rsync `--fuzzy --delete-after` on the compressed files only, (using an include or filter rule), then running rsync `--in-place` on the uncompressed main log file.

Handling the case of the uncompressed files which can be created on the first rotation (e.g. `/var/log/syslog -> /var/log/syslog.1`) might be more tricky, because I don't think `--fuzzy` will pick those up, but might be possible by comparing timestamps and moving manually prior to the rsync command?

azlux, 2018-03-21:
Sorry to be that long.
I read the doc of all you suggestion, but I have some doubt on the performances :
- `--inplace : WARNING: you should not use this option to update files that are being accessed by others.` The problem is that every program use the log files. I'm not sur what kind of side effect it's will make.

- `--fuzzy :  The current algorithm looks in the same directory as the destination file for either a file that has an identical size and modified-time, or a similarly-named file ` it's will not help to minimize writing because a logrotate change the name and the modified date file.

tim-seoss, 2018-03-21:
The `--inplace` warning is OK in this case, since rsync is only updating the files which are on the SD card (not the source files which are on the ram disk).  Nothing except rsync itself (as run by log2ram) will be updating the files on the SD card, so no problem.

log rotation does not change the modified time in my tests, e.g. 
```
1432709 Mar 21 14:59 syslog
1902886 Mar 21 00:10 syslog.1
 601886 Mar 20 00:05 syslog.2.gz
  49707 Mar 19 00:07 syslog.3.gz
 296755 Mar 18 00:08 syslog.4.gz
 106841 Mar 17 00:05 syslog.5.gz
 213485 Mar 16 00:06 syslog.6.gz
  58580 Mar 15 00:05 syslog.7.gz
```

after logrotate is run becomes:

```
    443 Mar 21 15:05 /var/log/syslog
1438756 Mar 21 15:05 /var/log/syslog.1
 161042 Mar 21 00:10 /var/log/syslog.2.gz
 601886 Mar 20 00:05 /var/log/syslog.3.gz
  49707 Mar 19 00:07 /var/log/syslog.4.gz
 296755 Mar 18 00:08 /var/log/syslog.5.gz
 106841 Mar 17 00:05 /var/log/syslog.6.gz
 213485 Mar 16 00:06 /var/log/syslog.7.gz
```

... for 5 of the files, all that's needed is a rename, but (without `--fuzzy`) rsync will end up deleting them, and renaming them.

But for both of these, don't just take my word for it, please do some tests .e.g using strace or iostat -d

azlux, 2019-07-23:
This discussion have move to #81. Especially because the --fuzzy command isn't the only one missing.

closed 2019-07-23 by azlux