azlux/log2ram #46: Suggestion: install uninstaller and remove the source folder then

thomas-mc-work, 2018-01-15:
On setup the `uninstaller.sh` should be copied to the target system as well. Then the downloaded and extracted archive could be removed. This way you always know where to find the uninstaller and are prevented from getting lost of it.

The target could be one of these:
- `/usr/local/log2ram-uninstaller.sh`
- `/usr/local/uninstall-log2ram.sh`

Also maybe the uninstaller script could be installed without execution permission to prevent uninstalling by accident.

The Readme could then be updated like this:

> ## Uninstall :(
> (Because sometime we need it)
> 
>     # chmod +x /usr/local/bin/uninstall-log2ram.sh && uninstall-log2ram.sh

azlux, 2018-02-07:
good point ! I will look at this suggestion.

MegaV0lt, 2018-02-08:
Bad when you have it downloaded via git clone to keep it actual

azlux, 2018-02-08:
@MegaV0lt keeping you updated by git clone don't install the product. So I don't think it's a bad idea.
And I think people who clone my project instead of following the Readme know what they do.

MegaV0lt, 2018-02-09:
I know. The point is when i want to update the installation (I know i have to uninsatll/install) it is bad when the uninstall removes my git-dir. I want to have the git to keep updatet setup files instead to go to some website and download the source manualy.

thomas-mc-work, 2018-02-09:
> when the uninstall removes my git-dir

Maybe I'm misunderstanding here. But I think the uninstaller only removes the installation, not the setup folder. Actually the whole thing is independent of the distribution of the installer.

azlux, 2018-02-16:
I've take care of you both comment.
Uninstaller is copy into /usr/local/bin for futur uninstall.
I've add the command to remove the folder of log2ram into the README. The script will not to do itself.

heikoh81, 2018-02-18:
On your GitHub frontpage, there is 1 error in the installation instructions.

`cd log2ram-master`
 is not needed for
`chmod +x log2ram-master/install.sh && sudo log2ram-master/install.sh`

Even more, cd-ing into the extracted directory produces an error.
This could confuse unexperienced users!

azlux, 2018-02-18:
https://github.com/azlux/log2ram/commit/644739c9250252ccbc927d84b977c9241daaa41a : fast fix for the `cd` error.
If I remove the `cd` , I need to change the path of all copy of the `install.sh` script.

heikoh81, 2018-02-25:
Just set up another Raspberry Pi (Jessie).
`chmod +x install.sh && sudo install.sh`
gives me
`root@raspTEST:~/log2ram-master# chmod +x install.sh && sudo install.sh
sudo: install.sh: command not found
`

Entering `./install.sh` works, however.

closed 2018-03-03 by azlux