azlux/log2ram #47: not working on my system

bernid, 2018-02-16:
Hi, 
Thanks for this piece of code.
I wanted to try to install this script on Ubuntu 17.10 Unfortunately the install program exited because there is no (on newest Ubuntu) /usr/local/bin. So I created /usr/local/bin and the installation has been completed. 
But still cannot run RAM Disk. I can see in the syslog:
```
Feb 16 19:47:20 akacja log2ram[808]: ERROR: RAM disk too small. Can't sync.
Feb 16 19:47:20 akacja log2ram[808]: mail: cannot send message: Process exited with a non-zero status
```
Maybe I should make symlink /usr/local/bin/log2ram to /usr/bin ?
Or maybe there are other ways to solve my issue?
TIA and regards, 
Regards

azlux, 2018-02-16:
RAM disk too small : Change the value SIZE of `/etc/log2ram.conf`
You can check the size of your log with `du -sh /var/log`

azlux, 2018-02-16:
That's strange than Ubuntu remove `/usr/local/bin` because `/usr/bin` is here for the packet manager.
Have you the folder `/usr/local/sbin` ?

bernid, 2018-02-16:
> RAM disk too small : Change the value SIZE of /etc/log2ram.conf
Thanks, my /var/log had 124M.. After cleaning has 74M. I set RAM disk SIZE=150M (have 8GB RAM)
And now log2ram seems to be working but at the same time when I look at my system SDD writing activity graph over time there is no change: still average 1.9M/s writing activity. I have to investigate this. 

>Have you the folder /usr/local/sbin ?
Yes I have (on my Ubuntu server 17.10)  but empty.

azlux, 2018-04-30:
Lot of system use  `/usr/local/bin/`, so I make the script creating the folder if not exist.

Thank

closed 2018-04-30 by azlux

azlux, 2019-12-19:
FYI https://github.com/azlux/log2ram/commit/13d0058086ff026b2c97723d248532f2237cfa23
Better install method, should work even if the folder isn't created.