azlux/log2ram #48: [Feature] OpenRC compatibility

MegaV0lt, 2018-02-18:
Is it possible to adapt log2 ram to OpenRC that is used on Gentoo Linux instead of Systemd?

azlux, 2018-02-19:
I think Gentoo have init.d system ?

MegaV0lt, 2018-02-19:
Yes, i have /etc/init.d... Sorry. At boottime on tehn screen i can see openrc starting. i think i was misunderstanding that.

azlux, 2018-02-19:
related to #16

azlux, 2018-02-19:
Are you familiar to use iniV ? I will allow init configuration, but I cannot test it here.

MegaV0lt, 2018-02-19:
No, sorry. Never heard about iniV

azlux, 2018-02-19:
Sorry the name of init.d script is sysvinit

MegaV0lt, 2018-02-19:
Well, minimal... Have seen already some but can not convert from systemd :-(
The Gentoo i used is Gen2VDR that is a distribution bundled with VDR and Kodi to have an Linux HTPC. Bas System is Gentoo.

The reqested Output:

```
root auf hdvdr01 am 19.02.2018 13:52 
[~] # cat /proc/1/comm
init
```

MegaV0lt, 2018-02-19:
I found this: https://github.com/akhilvij/systemd-to-sysvinit-converter
But i doubt it is much help

azlux, 2018-02-19:
@fugitive90 Maybe you can help with that. I see you have fork my project for FreeBSD use. We can open a branch for sysvinit compatibility.

fugitive90, 2018-04-04:
Hello everyone.
Sorry didn't noticed notification. @azlux yes, I planed to made for sysvinit and open-rc, but didn't got time to finish it. I'll see what I can do.

azlux, 2018-04-07:
@fugitive90 You can make changes into https://github.com/azlux/log2ram/tree/sysvinit
I see you have change many function, maybe we can create a functional branch for init.d system and add other changes later.

fugitive90, 2018-04-11:
@azlux Please find me on freenode, #devuan,#alpine-linux or #artix nick ``fugitive'' so we can discuss it.

MegaV0lt, 2018-06-01:
i tried to install om my VDR. But it fails:
> root auf hdvdr01 am 01.06.2018 13:30 
[~/log2ram-sysvinit] # bash ./install.sh
./install.sh: Zeile 14: systemctl: Kommando nicht gefunden.
##### Reboot to activate log2ram #####

Sailing74, 2018-06-01:
Does your machine run systemd? Looks like this is not the case.

MegaV0lt, 2018-06-01:
Is Gentoo Linux

> root auf hdvdr01 am 19.02.2018 13:52 
[~] # cat /proc/1/comm
init

MegaV0lt, 2018-06-01:
i think it is openrc-init
https://wiki.gentoo.org/wiki/Init_system

azlux, 2018-06-01:
@MegaV0lt Sorry I cannot make the init script for gentoo. I never use it. If you think you can do it, I will create a Branch for that.

karlh626, 2018-06-06:
I have several raspberry pis running Rasbian (Debian 7) which still uses init instead of systemd.  I don't know if a solution for init on Rasbian might benefit the Gentoo distro in any way?

MegaV0lt, 2018-06-07:
It is usinf init.d-Scripts. Example:

```
root auf hdvdr01 am 07.06.2018 08:31 
[~] # cat /etc/init.d/ntpd
#!/sbin/openrc-run
# Copyright 1999-2016 Gentoo Foundation
# Distributed under the terms of the GNU General Public License v2
# $Id$

description="ntpd - the network time protocol daemon"
pidfile="/var/run/ntpd.pid"
command="/usr/sbin/ntpd"
command_args="-p ${pidfile} ${NTPD_OPTS}"
start_stop_daemon_args="--pidfile ${pidfile}"

depend() {
	use net dns logger
	after ntp-client
}

start_pre() {
	if [ ! -f /etc/ntp.conf ] ; then
		eerror "Please create /etc/ntp.conf"
		return 1
	fi
	return 0
}

```
start-stop-deamion is also in some scripts

MegaV0lt, 2019-09-23:
I started a attempt to include detection of init-system and install/uninstall files acording to the init-system.
My first try looks good so far. You can take a look here: https://github.com/MegaV0lt/log2ram

fhoshino, 2020-05-11:
Is it just recent update broke openrc support?

MegaV0lt, 2020-05-11:
What update you mean?

fhoshino, 2020-05-11:
Never mind, I just copied the old open-rc script from another alpine box and it works now, just guessing recent update removed / broke open-rc support

cake2, 2024-12-26:
Ive been on this quest for a while now. There is a tutorial here for sysV init. https://energytalk.co.za/t/gx-device-logging-to-ram-tmpfs-log2ram/2130

This worked on my bananapi running latest Devuan (bookworm) However on a couple of odroids I tried to install it too failed, the big diffrence between the two is the odroids are using full disk encryption. The problem is in /etc/ram2log.conf is  PATH_DISK="/var/log". That is where I am stuck. When starting the system it hangs.

MegaV0lt, 2024-12-26:
Well, i am not using Gentoo anymore, sorry

cake2, 2024-12-26:
> Well, i am not using Gentoo anymore, sorry

Okay, :-)

Anybody elese who is interested, I hooked up serial port to see whats going on.
This is on a device using full disk encryption (cryptsetup, initramfs, dropbear ....)

`.....
cryptsetup: rootfs: set up successfully
e2fsck 1.47.0 (5-Feb-2023)
Superblock last mount time is in the future.
        (by less than a day, probably due to the hardware clock being incorrectly set)
Superblock last write time is in the future.
        (by less than a day, probably due to the hardware clock being incorrectly set)
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Pass 5: Checking group summary information
/dev/mapper/rootfs: 51300/3801088 files (0.1% non-contiguous), 796895/15199488 blocks
SELinux:  Could not open policy file <= /etc/selinux/targeted/policy/policy.33:  No such file or directory
INIT: version 3.06 booting
INIT: No inittab.d directory found
Using makefile-style concurrent boot in runlevel S.
Starting fake hwclock: loading system time.
Thu Dec 26 23:30:49 UTC 2024
Starting log2ram:
touch: cannot touch '/var/hdd.log/log2ram.test': Read-only file system
touch: cannot touch '/var/hdd.log/log2ram.test': Read-only file system
.....`

I think I can figure this out, I'll comment here once again if I get it working. Anybody that may be interested- like I said the above link works on latest devuan (bookworm--SysV init) without the disk encryption. cheers

cake2, 2024-12-27:
I got it, Ill post a tutorial over on devuan's forum for anyone interested.
I just had to edit (from the link I posted above) /etc/init.d/log2ram-init.sh
add $local_fs to # Required-Start

....
## BEGIN INIT INFO
# Provides:          log2ram
# Required-Start: $local_fs 
# Required-Stop:        
# Default-Start:     S
.....

It can probably be refined, but df -h shows log2ram :-)

melizasw, 2025-08-02:
Created PR #249 to add sysvinit scripts based on what @cake2 found and tweaked. These are the scripts that I have used for a Devuan package of log2ram - https://git.devuan.org/SteveM/log2ram