azlux/log2ram #50: fails to remove old version upon installation

nachoparker, 2018-04-23:
Hi,

When trying to install in a system with the old system ( `/var/log.hdd` ) it fails to remove it ( [this line](https://github.com/azlux/log2ram/blob/master/install.sh#L21) ) with

```
rm: cannot remove '/var/log.hdd': Device or resource busy
```

Probably needs to be umounted and rebooted first. Could create a PR if we agree on a way to do it.

azlux, 2018-04-23:
https://github.com/azlux/log2ram/commit/e41d5a0f23ca9c7a62f419412122b2960819aade
Here the procedure. Thank for pointing me that.

closed 2018-04-23 by azlux

azlux, 2018-04-23:
Adding this line can be enought.
```
systemctl -q is-active log2ram  && echo { echo "ERROR: log2ram still runnin. please do a \" sudo service log2ram stop \""; exit 1; }
```

nachoparker, 2018-04-23:
yes, that's exactly what I was thinking ;)

AlexanderPoleschuk, 2018-04-30:
To be more precise :)
"ERROR: log2ram service is still running. Please run \" sudo service log2ram stop \"" to stop it.