azlux/log2ram #51: Installed as described in rPi Stetch; /var/log/log2ram.log is empty

M-ax-G, 2018-05-05:
I installed log2ram on a freshly installed Stretch Lite on a Raspberry Pi 3B, yesterday evening.

\# [2018-05-05 22:28] maxg@rpi31 ~/downloads/log2ram-master $
`chmod +x install.sh && sudo ./install.sh
Created symlink /etc/systemd/system/sysinit.target.wants/log2ram.service → /etc/systemd/system/log2ram.service.`

`df -h` shows that /var/log has grown from:

`log2ram          40M  2.7M   38M   7% /var/log`
to
`log2ram          40M  3.5M   37M   9% /var/log`

checking the log with
`cat /var/log/log2ram.log`
returns noting; checking the file size of the log reveals 0 bytes:
`ls -l /var/log/log2ram.log`
`-rw-r--r-- 1 root root 0 May  5 22:31 /var/log/log2ram.log`

`ls -la /etc/cron.hourly/`
`-rwxr-xr-x  1 root root   44 May  5 22:29 log2ram`

`cat /etc/cron.hourly/log2ram`
`\#!/usr/bin/env sh`
`systemctl reload log2ram`

Any hints what I can further check or do to get log file entries?

azlux, 2018-05-05:
Hi,
If you haven't  change the configuration file, you use `cp` , so there are no log as soon as there is no error during the copy.
No new, good news. :D

M-ax-G, 2018-05-05:
Thanks... makes sense... only report if there is an issue; great. :)

closed 2018-05-05 by M-ax-G