azlux/log2ram #53: ZRAM or TMPFS

azlux, 2018-06-28:
For some usage, ZRAM look like a good alternative to TMPFS mount point.
We can save more data into a small part of the ram.

Not sure if it's a good idea.
If I add this option, it will be a option (disabled by default)

Feel free to answers this ticket to give me your opinion / idea. Or even more, make a pull request.

Az

d-a-v, 2019-03-04:
Do think this great idea could be reconsidered ?

azlux, 2019-03-04:
@d-a-v 
Related to #61 and #59 ? @StuartIanNaylor has start working on it.
I've never work with zram. So I need people to suggest and validate changes.

d-a-v, 2019-03-04:
Well yes, thanks for the pointer (I only jumped to explicit issue titles :)
 I'll give it a try asap!

azlux, 2019-03-15:
@d-a-v Have you test the last commit on the main branch,? zram is an option.
I'm not able to make test for now.

StuartIanNaylor, 2019-03-15:
Not really sure about tmpfs as in raspbian its all mem mounted already.