azlux/log2ram #54: Raspberry lite console autologin failed after install log2ram

wangweihong, 2018-08-20:
After I installed log2ram, console autologin fail. I have tried to config raspi-config again and again, it still ask me to enter password. After I stop and disable log2ram service, autologin works again. How can i fix this problem? Thanks for any help.

azlux, 2018-08-20:
Can i have the output of systemd-analyze plot 
And systemd-analyze critical-chain i suspect apache starting before rsyslog.
And please use [a paste bin](https://azlux.fr/bin) for the first output, it's a svg picture.
You can also check the /var/log.hdd/log2ram.log file if there are any error here.
I will check the next week.

Az

wangweihong, 2018-08-21:
@azlux Thanks for your reply. Directory '/var/log.hdd' doesn't exists, but '/var/hdd.log' does. And there is not logs in this dir.
Here is my `systemd-analyze plot`  **without** log2ram output:
https://azlux.fr/bin/?8b4f507581f09baf#1ikqr8T4hzaDqr7m9+1H5VLwOULW40TlJatUoRk5vBY=
`system-analyze critial-chain` is here:
```
i@raspberrypi:~ $ systemd-analyze critical-chain
The time after the unit is active or started is printed after the "@" character.
The time the unit takes to start is printed after the "+" character.

multi-user.target @8.616s
└─hciuart.service @4.071s +4.541s
  └─basic.target @4.061s
    └─sockets.target @4.053s
      └─triggerhappy.socket @4.046s
        └─sysinit.target @4.015s
          └─systemd-timesyncd.service @3.697s +315ms
            └─systemd-tmpfiles-setup.service @3.610s +56ms
              └─systemd-journal-flush.service @3.526s +78ms
                └─systemd-journald.service @3.393s +128ms
                  └─log2ram.service @2.802s +584ms
                    └─var-log.mount @2.879s
                      └─local-fs-pre.target @1.696s
                        └─keyboard-setup.service @1.034s +648ms
                          └─systemd-journald.socket @1.004s
                            └─-.slice @754ms
```

Here is my `systemd-analyze plot`  **with** log2ram output:
https://azlux.fr/bin/?b8171bb495ac66d9#PUYkvV8Xwknf8j8tjxfdmys7HNONZQV9ZTKMKzr6uko=

`system-analyze critial-chain` is here:
```
The time after the unit is active or started is printed after the "@" character.
The time the unit takes to start is printed after the "+" character.

multi-user.target @8.616s
└─hciuart.service @4.071s +4.541s
  └─basic.target @4.061s
    └─sockets.target @4.053s
      └─triggerhappy.socket @4.046s
        └─sysinit.target @4.015s
          └─systemd-timesyncd.service @3.697s +315ms
            └─systemd-tmpfiles-setup.service @3.610s +56ms
              └─systemd-journal-flush.service @3.526s +78ms
                └─systemd-journald.service @3.393s +128ms
                  └─log2ram.service @2.802s +584ms
                    └─var-log.mount @2.879s
                      └─local-fs-pre.target @1.696s
                        └─keyboard-setup.service @1.034s +648ms
                          └─systemd-journald.socket @1.004s
                            └─-.slice @754ms

```

azlux, 2018-08-28:
Hi,
I now have time to work on it. Badly, the pastebin are expired (the default value is one week).

wangweihong, 2018-08-31:
@azlux  Hi, azlux. Sorry for late reply,  I find `autlogin@.service` doesn't enable after system boots. After I run`systemctl enable autologin@.service`, it seems autologin work fine now. Still don't know why it happen, I will try to figure it out. THX for your help.

azlux, 2018-11-13:
Have you found something ?
Can I close this ticket ? Do you need more help ?

wangweihong, 2018-11-20:
Sorry, I forgot to close this issue.

closed 2018-11-20 by wangweihong