azlux/log2ram #57: Issue with selinux?

kingfisher77, 2018-12-23:
Hi, i am on a fedora workstation current release and thought it is a good idea to use log2ram (ssd as boot disk).
On a fresh install of log2ram i experience problems while booting. sssd, auditdm, firewalld, gdm a.s.o can not write to log because of permission denied. Permissions are default. An restorecon -rv /varlog/ (and /var/log/hdd.log) fixes selinux permissions. The daemons can start after that.

What do i miss?

azlux, 2018-12-24:
Hi,
Sorry, I've never work with SELinux.
I'm not sure I can help.
Do you have any error into /var/hdd.log/log2ram.log ?

Az

azlux, 2019-01-02:
Have you the errors you can sees in audit (aka AVC denials)
The command ausearch -m avc -i will show these.

kingfisher77, 2019-01-03:
I am on vacation until mid of january. Will come back to you later.

xupefei, 2019-05-02:
Hi all,

The solution is very simple here: when restoring `hdd.log` to `log` after each boot, just tell the `cp` command to copy also extended attributes. This is done by changing **Lines 23 and 43** of `log2ram` from 

```
cp -rfup <other stuff>
```

to 

```
cp -rfup --preserve=context <other stuff>
```

Note: you need to change both Lines 23 and 43.

The `rsync` command works without any modification because it already has `-X` flag that preserves the SELinux attributes.

Ref: https://linuxconfig.org/copy-a-file-while-preserving-selinux-context-on-linux

kingfisher77, 2019-05-02:
Great! Interesting - thank you!

closed 2020-05-04 by azlux