azlux/log2ram #58: log2ram.service failed (timeout)

EPiC-APOC, 2019-02-10:
Linux RasPiHole 4.14.97-v7+ #1197 SMP Mon Feb 4 21:13:29 GMT 2019 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sun Feb 10 14:02:32 2019
pi@RasPiHole:~ $ `cd log2ram-master`
pi@RasPiHole:~/log2ram-master $ `systemctl status log2ram.service`
● log2ram.service - Log2Ram
   Loaded: loaded (/etc/systemd/system/log2ram.service; enabled; vendor preset: enabled)
   Active: failed (Result: timeout) since Sun 2019-02-10 14:01:29 CET; 3min 23s ago
  Process: 217 ExecStart=/usr/local/bin/log2ram start (code=killed, signal=TERM)
 Main PID: 217 (code=killed, signal=TERM)

Feb 10 14:01:29 RasPiHole log2ram[217]: [58B blob data]
Feb 10 14:01:29 RasPiHole log2ram[217]: [58B blob data]
Feb 10 14:01:29 RasPiHole log2ram[217]: [59B blob data]
Feb 10 14:01:29 RasPiHole log2ram[217]: mount: wrong fs type, bad option, bad superblock on log2ram,
Feb 10 14:01:29 RasPiHole log2ram[217]:        missing codepage or helper program, or other error
Feb 10 14:01:29 RasPiHole log2ram[217]:        In some cases useful info is found in syslog - try
Feb 10 14:01:29 RasPiHole log2ram[217]:        dmesg | tail or so.
Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.
 
/log2ram-master $  `dmesg | tail`
[  125.479972] systemd[1]: log2ram.service: Main process exited, code=killed, status=15/TERM
[  125.480821] systemd[1]: Failed to start Log2Ram.
[  125.481205] systemd[1]: log2ram.service: Unit entered failed state.
[  125.481248] systemd[1]: log2ram.service: Failed with result 'timeout'.
[  125.486038] systemd[1]: Starting Journal Service...
[  125.594830] systemd[1]: Started Journal Service.
[  125.643757] systemd-journald[2469]: Received request to flush runtime journal from PID 1
[  126.593789] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[  126.621515] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[  128.228864] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xC5E1

azlux, 2019-02-10:
Hi,
Can you give me the output of `sudo fdisk -l` ?

Az

EPiC-APOC, 2019-02-10:
Weirdest thing, reinstalled again and now it seemed to work, did try this a few times for posting here, but without any result. Even on a fresh Raspbian setup/install.

I will try some reboots and see if it sticks, my results for sudo fdisk -1 are at this moment (and with your script working so it seems) :
pi@RasPiHole:~ $ sudo fdisk -l
Disk /dev/ram0: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram1: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram2: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram3: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram4: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram5: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram6: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram7: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram8: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram9: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram10: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram11: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram12: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram13: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram14: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram15: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/mmcblk0: 29.8 GiB, 32009355264 bytes, 62518272 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xe9ba3771

Device         Boot Start      End  Sectors  Size Id Type
/dev/mmcblk0p1       8192    98045    89854 43.9M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      98304 62518271 62419968 29.8G 83 Linux
pi@RasPiHole:~ $

EPiC-APOC, 2019-02-10:
Oops 1 reboot and it's gone again :'(

pi@RasPiHole:~ $ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        30G  1.4G   27G   5% /
devtmpfs        484M     0  484M   0% /dev
tmpfs           489M  248K  489M   1% /dev/shm
tmpfs           489M  6.5M  482M   2% /run
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           489M     0  489M   0% /sys/fs/cgroup
/dev/mmcblk0p1   44M   22M   22M  51% /boot
tmpfs            98M     0   98M   0% /run/user/999
tmpfs            98M     0   98M   0% /run/user/1000

pi@RasPiHole:~ $ sudo fdisk -l
Disk /dev/ram0: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram1: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram2: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram3: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram4: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram5: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram6: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram7: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram8: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram9: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram10: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram11: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram12: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram13: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram14: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram15: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/mmcblk0: 29.8 GiB, 32009355264 bytes, 62518272 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xe9ba3771

Device         Boot Start      End  Sectors  Size Id Type
/dev/mmcblk0p1       8192    98045    89854 43.9M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      98304 62518271 62419968 29.8G 83 Linux
pi@RasPiHole:~ $

azlux, 2019-02-11:
Wait. You are on raspberry and you have many tmpfs reserving ram space. To you think there are enought RAM space available for log2ram (depending of uour configuration)

Can you check with `free -h` ?

EPiC-APOC, 2019-02-11:
I did not make these ramdisks by the way, could they be leftovers from log2ram uninstall's perhaps?
...and I'll check `free -h` later on for you, not near the Raspberry Pi2B+ at this moment...

azlux, 2019-02-11:
It's not possible.
Log2Ram do not create tmpfs is the service cannot start.
Do you use the raspbian lite ?

EPiC-APOC, 2019-02-11:
Yes I use the latest (up2date) **Raspbian Stretch Lite** (https://www.raspberrypi.org/downloads/raspbian/)

EPiC-APOC, 2019-02-11:
```
pi@RasPiHole:~ $ free -h
              total        used        free      shared  buff/cache   available
Mem:           976M        126M        570M         15M        279M        785M
Swap:           99M          0B         99M
pi@RasPiHole:~ $
```
I did uninstall log2ram at this moment, or did you need this result with log2ram "enabled"
(probably not since it does not work for me at the moment, sometimes it does, really weird)

azlux, 2019-02-11:
Strange.
Can I have the output of `mount` ?

EPiC-APOC, 2019-02-12:
**mount**

```
pi@RasPiHole:~ $ mount
/dev/mmcblk0p2 on / type ext4 (rw,noatime,data=ordered)
devtmpfs on /dev type devtmpfs (rw,relatime,size=495484k,nr_inodes=123871,mode=755)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,mode=755)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,release_agent=/lib/systemd/systemd-cgroups-agent,name=systemd)
cgroup on /sys/fs/cgroup/freezer type cgroup (rw,nosuid,nodev,noexec,relatime,freezer)
cgroup on /sys/fs/cgroup/cpuset type cgroup (rw,nosuid,nodev,noexec,relatime,cpuset)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,nosuid,nodev,noexec,relatime,blkio)
cgroup on /sys/fs/cgroup/cpu,cpuacct type cgroup (rw,nosuid,nodev,noexec,relatime,cpu,cpuacct)
cgroup on /sys/fs/cgroup/net_cls type cgroup (rw,nosuid,nodev,noexec,relatime,net_cls)
cgroup on /sys/fs/cgroup/devices type cgroup (rw,nosuid,nodev,noexec,relatime,devices)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=29,pgrp=1,timeout=0,minproto=5,maxproto=5,direct)
mqueue on /dev/mqueue type mqueue (rw,relatime)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
sunrpc on /run/rpc_pipefs type rpc_pipefs (rw,relatime)
configfs on /sys/kernel/config type configfs (rw,relatime)
/dev/mmcblk0p1 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
tmpfs on /run/user/999 type tmpfs (rw,nosuid,nodev,relatime,size=100016k,mode=700,uid=999,gid=996)
tmpfs on /run/user/1000 type tmpfs (rw,nosuid,nodev,relatime,size=100016k,mode=700,uid=1000,gid=1000)
pi@RasPiHole:~ $
```

azlux, 2019-02-13:
Your case is curious.
First time I see this issue.
Can you enable log2ram again ?
I also need the ouput of :
- `findmnt --verify`
- `cat /etc/log2ram.conf | grep SIZE`
- `free -h`
- `cat /etc/fstab`
- `systemd-analyze plot ` - please use  [a paste bin](https://azlux.fr/bin) for this one. Lot of output, it's a svg picture.  
- `systemd-analyze critical-chain`
- `whereis mount`
- `mount -V`

Can you try to manually mount a folder to test if you have the same error, and same `dmesg` ? Remove the folder `/var/hdd.log/` and use the following command Here the command:
```
mount --bind /var/log/ /var/hdd.log/
mount --make-private /var/hdd.log/
mount -t tmpfs -o nosuid,noexec,nodev,mode=0755,size="40M" log2ram /var/log/
```

EPiC-APOC, 2019-02-17:
findmnt --verify
```
/boot
   [W] cannot detect on-disk filesystem type
/
   [W] cannot detect on-disk filesystem type

0 parse errors, 0 errors, 2 warnings
```
sudo findmnt --verify : `Success, no errors or warnings detected`

cat /etc/log2ram.conf | grep SIZE
```
SIZE=40M
```
free -h
```
              total        used        free      shared  buff/cache   available
Mem:           976M        118M        738M         15M        119M        795M
Swap:           99M          0B         99M
```
cat /etc/fstab
```
proc                  /proc           proc    defaults          0       0
PARTUUID=e9ba3771-01  /boot           vfat    defaults          0       2
PARTUUID=e9ba3771-02  /               ext4    defaults,noatime  0       1
# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that
```
`systemd-analyze plot : making a paste bin later`

systemd-analyze critical-chain
```
The time after the unit is active or started is printed after the "@" character.
The time the unit takes to start is printed after the "+" character.

graphical.target @1min 31.004s
└─multi-user.target @1min 31.003s
  └─pihole-FTL.service @3.593s +12.052s
    └─basic.target @3.191s
      └─sockets.target @3.189s
        └─avahi-daemon.socket @3.186s
          └─sysinit.target @3.176s
            └─systemd-timesyncd.service @2.882s +291ms
              └─systemd-tmpfiles-setup.service @2.815s +49ms
                └─systemd-journal-flush.service @2.756s +53ms
                  └─systemd-journald.service @2.629s +120ms
                    └─syslog.socket @933ms
                      └─-.mount @752ms
                        └─system.slice @879ms
                          └─-.slice @752ms
```
whereis mount
```
mount: /bin/mount /sbin/mount.nfs4 /sbin/mount.nfs /sbin/mount.cifs /usr/share/man/man8/mount.8.gz /usr/share/man/man2/mount.2.gz
```
mount -V
```
mount from util-linux 2.29.2 (libmount 2.29.2: selinux, btrfs, assert, debug)
```

EPiC-APOC, 2019-02-17:
`systemd-analyze plot ` - please use  [a paste bin](https://azlux.fr/bin) for this one. Lot of output, it's a svg picture:
https://azlux.fr/bin/?5ddfd474368b7e5e#CgjswMxJmxkLef3NcaaYOK4UBKrxvh8R+fz9iEEybFw=

EPiC-APOC, 2019-02-17:
`sudo rm -r /var/hdd.log/`

mount --bind /var/log/ /var/hdd.log/
```
mount: only root can use "--bind" option
```
**sudo** mount --bind /var/log/ /var/hdd.log/
```
mount: mount point /var/hdd.log/ does not exist
```
**sudo** mkdir /var/hdd.log/
**sudo** mount --bind /var/log/ /var/hdd.log/
**sudo** mount --make-private /var/hdd.log/
**sudo** mount -t tmpfs -o nosuid,noexec,nodev,mode=0755,size="40M" log2ram /var/log/

EPiC-APOC, 2019-02-21:
The other mentioned pastebin is almost expired (2 days left), I cloned that one for an additional 1 month here:

https://azlux.fr/bin/?2bc796ad1ca5abef#vB0a4e3gJDHruMUMv0b0+H+MWMbApT5jCdZ0oxVZmsw=

azlux, 2019-02-21:
Thank, I'm out of time now.
I will check next week. Thank for all the informations.

EPiC-APOC, 2019-02-23:
> Thank, I'm out of time now.
> I will check next week. Thank for all the informations.

fair enough, no problem, I appreciate the effort of course and there's no hurry.
(just really curious what seems to be troubling, I can't get my head around it lol #annoying haha)

azlux, 2019-03-21:
Hi,
I see 2 things.
1. You have NFS, the client start in the same time as log2ram.
Where is your nfs mount point ? (If exist)
2. I see nothing else :'( , I will create a debug branch and add debug option on mount part.

azlux, 2019-03-21:
Can you try the install with the debug branch ?

Use `curl -Lo log2ram.tar.gz https://github.com/azlux/log2ram/archive/debug.zip` is first install step.

The journalctl and `service log2ram status` will output lot more information

azlux, 2019-04-29:
@EPiC-APOC 
I've reinstall log2ram on a fresh raspbian lite OS without any issue yesterday.

Can you try the debug branch ?

mclaser, 2019-10-18:
I believe I may have this same issue on my RPi.  The service appears to be running but it looks the ramdisk hasn't been established as it's too small.  Nothing in dh -h or mount.  I've had various projects on this RPi for the past few years and Pi-hole is currently running, apparently correctly.  I'm not sure if it's Lite or not.  It sounds like a clean install would resolve this but it would be great if I didn't have to do that.  Maybe I can help you figure out the issue and I can learn a little along the way ;)

```
uname -a
Linux rp2-ph 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux

cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
VERSION_CODENAME=stretch
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

 service log2ram status
 log2ram.service - Log2Ram
   Loaded: loaded (/etc/systemd/system/log2ram.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2019-10-18 13:39:08 PDT; 22min ago
  Process: 253 ExecStart=/usr/local/bin/log2ram start (code=exited, status=1/FAILURE)
 Main PID: 253 (code=exited, status=1/FAILURE)

Oct 18 13:39:08 rp2-ph log2ram[253]: ERROR: RAM disk too small. Can't sync.
Oct 18 13:39:08 rp2-ph log2ram[253]: /usr/local/bin/log2ram: 35: /usr/local/bin/log2ram: mail: not f
Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.

free -h
              total        used        free      shared  buff/cache   available
Mem:           926M         52M        763M         15M        110M        809M
Swap:           99M          0B         99M
pi@rp2-ph:~ $ cat /etc/log2ram.conf | grep SIZE
SIZE=40M
# LOG_DISK_SIZE is the uncompressed disk size. Note zram uses about 0.1% of the size of the disk when not in use
# LOG_DISK_SIZE is expected compression ratio of alg chosen multiplied by log SIZE
LOG_DISK_SIZE=100M

cat /etc/fstab
proc            /proc           proc    defaults          0       0
PARTUUID=c46ec333-01  /boot           vfat    defaults          0       2
PARTUUID=c46ec333-02  /               ext4    defaults,noatime  0       1
# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that

sudo fdisk -l
Disk /dev/ram0: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram1: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram2: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram3: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram4: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
[... 43 lines cut ...]
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram14: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram15: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/mmcblk0: 7.4 GiB, 7948206080 bytes, 15523840 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xc46ec333

Device         Boot Start      End  Sectors  Size Id Type
/dev/mmcblk0p1       8192    93813    85622 41.8M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      94208 15523839 15429632  7.4G 83 Linux
```

azlux, 2019-10-18:
@mclaser your issue isn't this one. Please check https://github.com/azlux/log2ram/issues/90

MegaV0lt, 2020-02-10:
I installed today and have timeout at boot.
No idea what goes wrong.
I have 4 GB RAM
dmesg: https://www.dropbox.com/s/calyxk6g0fqy8q5/dmesg.log?dl=0
Stranbge is that df shows a mounted log2ram

[~] # df -h
Dateisystem    Größe Benutzt Verf. Verw% Eingehängt auf
udev            1,9G       0  1,9G    0% /dev
tmpfs           386M    7,6M  379M    2% /run
/dev/sda2       220G     55G  155G   27% /
tmpfs           1,9G    8,0K  1,9G    1% /dev/shm
tmpfs           5,0M       0  5,0M    0% /run/lock
tmpfs           1,9G       0  1,9G    0% /sys/fs/cgroup
tmpfs           1,9G     52K  1,9G    1% /tmp
log2ram         150M     31M  120M   21% /var/log
/dev/sdb1       1,8T    592G  1,2T   34% /mnt/ST32000542AS
tmpfs           386M       0  386M    0% /run/user/0

closed 2020-05-04 by azlux