azlux/log2ram #59: Not an Issue

StuartIanNaylor, 2019-02-26:
It would be cool if Log2Ram could use a Zram disk.
Its likely that many if using Log2Ram prob have Zram also.

Just a suggestion to make ram footprint tiny whilst increasing log size.

StuartIanNaylor, 2019-03-01:
https://github.com/StuartIanNaylor/log2ram/tree/log2zram

Dunno if you want to grab it or not but please do if you think its valid

azlux, 2019-03-01:
Hi.
Can you do a push request ? I have some question on your code.
It's look like very interesting. 

Thank
Az

StuartIanNaylor, 2019-03-01:
You just need to test if any zram devices already exist.

`	if modinfo zram | grep -q 'zram_num_devices:' 2>/dev/null; then
		MODPROBE_ARGS="zram_num_devices=${zram_num_devices}"
	elif
		modinfo zram | grep -q ' num_devices:' 2>/dev/null; then
		MODPROBE_ARGS="num_devices=${zram_num_devices}"
	else
		exit 1
	fi
	modprobe zram $MODPROBE_ARGS`

Create the drive
`	#Add log2ram zram drive
	SIZE=$((SIZE * 1024 * 1024))
	if [ "$ZL2R" = true ]; then
		echo ${COMP_ALG} > /sys/block/zram${BIG_CORES}/comp_algorithm
		echo ${SIZE} > /sys/block/zram${BIG_CORES}/disksize
		mke2fs -t ext4 -O ^has_journal,extent,^uninit_bg /dev/zram${BIG_CORES}
		tune2fs -o journal_data_writeback  /dev/zram${BIG_CORES}`

Dunno about options as have been wonderig also the block really should be as small as possible but anything lower than the default 4096 just seemed to cause an error.

Then mount it.

`      if [ "$ZLTG" = true ]; then
       	mount -t ext4 -o nosuid,noexec,nodev,user=log2ram,data=writeback,noatime,nodiratime /dev/zram$BIG_CORES $RAM_LOG/
      else
      	mount -t tmpfs -o nosuid,noexec,nodev,mode=0755,size="$SIZE" log2ram $RAM_LOG/
      fi`

PS your uninstall misses a umount /var/hdd.log so the drive never gets deleted

if [ -d /var/hdd.log ]; then
    rm -r /var/hdd.log
  fi

azlux, 2019-03-13:
I close this issue because I've merge the changes #66 

`/var/hdd.log` isn't a mount point. So removing it is good.

closed 2019-03-13 by azlux