azlux/log2ram #64: How many writes does the hourly sync cause?

StuartIanNaylor, 2019-03-12:
The service reload via the hourly cron job causes sync to disk on each hour

There is an absence of any check to drive usage and from the cp command the whole $RAM_LOG is written out to HDD.
Isn't this likely to cause vastly more HDD writes than just appending to normal logs in many situations?

It completely lacks any form of control, probably should run more frequently but with logical tests if log usage necessitates a sync, but just to write out everything each hour?   
`syncToDisk () {
    isSafe

    if [ "$USE_RSYNC" = true ]; then
        rsync -aXWv --delete --links $RAM_LOG/ $HDD_LOG/ 2>&1 | $LOG_OUTPUT
    else
        cp -rfup $RAM_LOG/ -T $HDD_LOG/ 2>&1 | $LOG_OUTPUT
    fi
}`

azlux, 2019-03-12:
This project has been made for SD card. In case of HDD, your aren't limited to write cycle.
With a SD card, Less you write, better it is. Even if you have more data to write.

StuartIanNaylor, 2019-03-12:
HDD as apposed to Ram irrespective of if media is SD, Nand or Emmc still doesn't change the fact that writing out complete logs on any change every hour, especially as those logs increase in size, means yes frequency of write is lowered, but the write size is often huge in comparison.

It is dependent on log size but with the application of the sphere of log2ram of the general Pi & clone world of Arm SBC computers its debatable if L2R makes much difference in frequency in comparison to write Size.
I am not hear to debate the merits of block wear and the above was my first comment on the main question of why does it write every hour when its completely not needed.

The ramlog only needs to be written to hdd.log as it is programmatically named on stop.
I commented further in https://github.com/azlux/log2ram/issues/65 as also ramlog doesn't prune and holds pointless old logs in memory which again with the oldlog directive of logrotate can be easily partitioned and moved when total ramlog usage is at a certain level.

Reducing hourly writes by sometimes as little as 3 or 4 but increasing data write size upwards of a 1,000 is likely no better and maybe even worse.

closed 2019-04-09 by azlux