azlux/log2ram #72: systemd interaction

djh42, 2019-03-26:
I asked some questions related to log2ram on the systemd-devel mailing list. The thread starts at https://lists.freedesktop.org/archives/systemd-devel/2019-March/042312.html and there are two main suggestions:

(1) I was asking about how to allow journald to continue logging to the SD card. journald is intelligent about reducing writes but will try its best to write emergency messages during a crash, so these want to be written directly to persistent store, not buffered by log2ram. The suggestion by Colin Guthrie is to use a bind mount. This would be a nice feature to add.

(2) I mentioned the Before=systemd-journald.service dependency and Leonard Poettering advised that it is better to use Before=systemd-journal-flush.service

I also notice the Before=apache2.service dependency and have a question about it. Why is it there? Is it just one example of a service that must not be started until the RAM log directories are prepared? What about other services that write logs, such as redis or mysql? Does the service file need customising on every machine where it is run or is there some other technique could be used?

azlux, 2019-04-09:
Hi @djh42 .

I will try to answers all your questions:

1. Log2ram already use a bind mount.
2. I didn't know about the systemd-journal-flush.service, I will check that. Is there are any conflict to use it the way I've done it ?
3. Apache2 was an old example when you have others services. Since few mounth, Log2Ram start before the journal, It's now useless. Almost every service start after the journal service.

Az

djh42, 2019-04-09:
>  Log2ram already use a bind mount.
For /var/log/journal? To place it on persistent storage.

>  Is there are any conflict to use it the way I've done it ?
Not as far as I know, but I'm no expert.

> Apache2 was an old example when you have others services. Since few mounth, Log2Ram start before the journal, It's now useless. Almost every service start after the journal service.
I'm not sure what you mean here. You mean the before=apache2 is useless? If so, I think that's good news. :)

azlux, 2019-04-09:
Your right, `/var/log/journal` isn't on Disk. But I don't like place it on persistent storage.
My goal is to avoid SDcart write, and I don't know how journald manage that.

Barabba11, 2019-04-09:
May I add here a question?
Do ram2log write enterely the ram log folder into the log.hdd folder on sd every hour? I mean, does it overwrite all files including the pnes that aren't untuched, or only the ones that have more recent modified time (then the last hour, I suppose). Shortly, does ram2log reduces the writes on SD avoiding to rewrite files that haven't been modified? 
Thank you

azlux, 2019-04-09:
Hi,
If you check this line https://github.com/azlux/log2ram/blob/master/log2ram#L21
You will see I only sync the modified file.

The main issue with SDcart is to have constant write, file are always open into the system.

You can move the cron from hourly to daily if you want (Or remove it, for no cron write, data will be write on Raspberry stop/start)

djh42, 2019-04-10:
> My goal is to avoid SDcart write, and I don't know how journald manage that

It already has a sensible mechanism to minimise writes, whilst also guaranteeing that as much as possible makes it to disk in the event of a system crash. That's why I believe /var/log/journal should be bound to persistent storage. Full details are at https://www.freedesktop.org/software/systemd/man/journald.conf.html but basically it writes to disk at a configurable period - SyncIntervalSec - that defaults to 5 minutes but can of course be adjusted by the user.

arendtio, 2019-04-12:
@azlux Actually, I just came across this project and was wondering what this is all about. I mean, I know that people are trying to avoid disk-writes on their SSDs and SD-Cards, but AFAIK that is completely possible with `/etc/systemd/journald.conf` (e.g. `Storage=volatile`).

djh42, 2019-04-12:
> Actually, I just came across this project and was wondering what this is all about. I mean, I know that people are trying to avoid disk-writes on their SSDs and SD-Cards, but AFAIK that is completely possible with /etc/systemd/journald.conf (e.g. Storage=volatile).

It's about the opposite. The purpose of the project as a whole is to stop all applications, not just journald, from logging to SD card. The purpose of this discussion is to request an exception so that journald DOES log to SD card, since it understands how to minimise writes to the card anyway, and since it can be important to have a persistent log of the last messages in a system crash.

closed 2020-02-05 by azlux