azlux/log2ram #73: still see entries in syslog file

Barabba11, 2019-04-04:
Hi mates, thank you a lot for this program!
I'm not an expert user of linux so I can be wrong here, but I noticed that system keeps writing entries on syslog file into var/log folder.. I mean before the "hour" when I suppose log2ram dump all lines into files. 
For this reason I'm not sure it works correctly, but on syslog I have every hour the lines:
Apr  5 00:17:01 raspberrypi systemd[1]: Reloading Log2Ram.
Apr  5 00:17:01 raspberrypi systemd[1]: Reloaded Log2Ram.
For example

I would like to reduce as minimum as possible the writes on SD, a friend suggested me this:
$ cat /etc/fstab
proc            /proc           proc    defaults          0       0
PARTUUID=7b4d7c24-01  /boot           vfat    defaults          0       2
PARTUUID=7b4d7c24-02  /               ext4    defaults,noatime,commit=600,errors=remount-ro  0       1
tmpfs /tmp tmpfs defaults,noatime,nosuid 0 0
tmpfs /var/log tmpfs defaults,noatime,nosuid,size=16m 0 0

but I think his solution doesn't dump anything to file, just move all folder log into RAM, and I don't want to lose log files after rebbot/shutdown/power loss.
May you please help me clear this? Thank you a lot!

StuartIanNaylor, 2019-04-05:
The Reloading Log2Ram isn't really necessary as its debatable on a complete crash if anything up to an hour is lost then what use is the hour preceeding.
You can use the service reload function at anytime to sync to persistent bind directory.

So remove the cron job as that is all its doing and really Reloading Log2Ram means it is working.

Barabba11, 2019-04-06:
thank you for your kind reply and support! I?m not a linux expert so I'm trying to figure out correctly what you wrote.. 
the main question was about the fule syslog which keeps writing on the SD, and I expected to be sync only after reload, why? 
Maybe log2ram is logging on RAM some logs but not all the log folder?
About reloading I'm fine with that, as well as lose the last hour of activity. Thank you

StuartIanNaylor, 2019-04-06:
Apols basically what log2ram does because you can not mount zram ontop of an existing directory.

Firstly it creates a bind mount so that /var/log can be moved elsewhere then you can create a /var/log in zram, but its empty.
So it copies all from the original /var/log now at a bind mount on start.
zram is completely volatile and non persistent and log2ram writes via a cron job every hour to update any changes and also on stop.

I sort of agree the hourly write is sort of pointless in terms of SD write so in /etc/cron.hourly delete the log2ram entry and no more hourly writes but sync will still happen on start or stop.

There are a few things with the service and method of log2ram that IMHO are mweh.
Have a look at https://github.com/StuartIanNaylor/zram-config as it sound like it might suit your needs better,

Barabba11, 2019-04-06:
sorry maybe I expressed myself wrongly, I'm fine with hourly writes on sd, I don't want to chenage them.
The problem is that the file var/log/syslog is keep written when an event is coming, I don't need to wait one hour to see the entry, I open anytime the file and see the new entries like before.
If you said that the entirely log folder is created on ram and updated only there, I don't know how this is possible.
I hope you understood my point, thanks for help!

azlux, 2019-04-06:
Hi
It's because with log2ram, the folder /var/log is a RAM folder (tmpfs to be exact).

The syslog system don't care I have change the type of this folder and continue to write all log as a simple directory.

Hope it help.
Az

Barabba11, 2019-04-06:
so if I understood right the path /var/log is now on RAM, and how and where can I see the real var/log on the sd? 
thank you

azlux, 2019-04-06:
The data on Disk is on /var/hdd.log/

Barabba11, 2019-04-06:
very interesting, perfect! thanks

closed 2019-04-06 by Barabba11