azlux/log2ram #74: Use a OverlayFS to negate need to syncFromDisk

StuartIanNaylor, 2019-04-05:
You can use an overlayFS.

Just mount tmpfs on a created directory
```
mkdir -p /opt/zram$RAM_DEV
mount --verbose --types ext4 -o nosuid,noexec,nodev /dev/zram$RAM_DEV /opt/zram$RAM_DEV/ 
```
I am using zram and just create a dir same a the dev name and mount there.

`mkdir -p /opt/zram$RAM_DEV/upper /opt/zram$RAM_DEV/workdir $ZRAM_DIR`
$ZRAM_DIR would be /var/log but just make the upper and workdir in zram create /log/var after it has been bind mounted.
Then create your OverlayFS mount which negates any need to syncFromDisk as you have a OverlayFS with zram upper on top of the bind mount lower.

`mount -t overlay -o lowerdir=$BIND_DIR,upperdir=/opt/zram$RAM_DEV/upper,workdir=/opt/zram$RAM_DEV/workdir overlay $ZRAM_DIR`

The code is from https://github.com/StuartIanNaylor/zram-config/tree/OverlayFS where I have a ztab creating all zram dirs / swaps and also use OverlayFs to negate the need for syncFromDisk that might cause delay on large directories.
syncToDisk only writes updated file changes so should be much less and is still in operation.

StuartIanNaylor, 2019-04-08:
azlux have a look at the commit dated today zram-config with overlay merge function.
https://github.com/StuartIanNaylor/zram-config

azlux, 2019-04-08:
@StuartIanNaylor If you have fix/improvement, please Pull requests it.
I don't have zram configurated on my device for now.

StuartIanNaylor, 2019-04-09:
Yeah that is sort of a cul-de-sac, apols.

closed 2019-08-19 by azlux