azlux/log2ram #75: Ext4 journaling isn't helpful here.

ClashTheBunny, 2019-04-08:
You create a filesystem with ext4 on the zramfs. It would be better to turn off journaling here. Journaling only helps if you can get back to the data after a crash, and there is no hope of that on a ram disk. You could either turn off journaling or switch to a bare bones filesystem like ext2.
https://github.com/azlux/log2ram/blob/07b3599823669a93870f946c7591f885aa370e2e/log2ram#L64

azlux, 2019-04-09:
You have a point here.

Ping @StuartIanNaylor

StuartIanNaylor, 2019-04-17:
Just been running some benches and -O ^has_journal seems in overall size and performance the same as defaults and seems to make no difference in operation.
In fact with zram any 'perf' option on mount of mkfs seems to make zero difference.

closed 2019-04-29 by azlux