azlux/log2ram #78: Disk size for tmpfs is too small

StuartIanNaylor, 2019-04-25:
Its too small and pointlessly too small as tmpfs is backed by swap so generally idle logs will be pushed out to swap by the system because its tmpfs and the disk size contains both and the defaults are too small.
My knowledge of system memory tmpfs and zram has sort of increased and its taken me round house that actually you don't need to use zram as a backing store as you can simply enable zram swap and the system will push out when idle.
Zram would be better if the writeback methods where better as all live logs would be compressed with minimal cpu overhead.
Unlike tmpfs the zram writeback methods are not automatic at least from what I gather. After assigning a swap partition specifically allocated to each zram drive you mark all as idle and on write that attribute is cleared but the method to writeback all still marked as idle is manual.
Also being partition based is also a pain with writeback and not insurmountable but just a pain as they need to be created and controlled.

If its KISS then using tmpfs with a zram or zswap system service means with zram you only need a single swap partition but gives you the choice of zswap that I think can be file based and the writeback is all automated.
Its quite easy to implement with zram but you actually have to implement a writeback service.

So to cut a long story short is the current tmpfs allocation is probably too restrictively small. As that size can be a combination of mem & swap and actually doesn't mean all that size will be constantly allocated. So some mention that tmpfs will use swap for idle pages and logs so that simply using a zram or zswap service with log2ram is actually a really good KISS idea.

Sorry for going round in circles but unless a zram writeback service is in place and a dedicated swap partition then its prob just as good and simpler to use tmpfs with a zswap service.

azlux, 2019-04-29:
If you use the swap you may have issue on your system.
I know default swappiness is quite hight. But for the log2ram goal, the best is to set the swappiness around 10.

I think I will add that on the Doc.

What you describe can actually go against the project in the case of swap. But I don't think I should add swap management in log2ram because it seems that zram needs improvement before it can be managed smoothly.

StuartIanNaylor, 2019-04-29:
The best to set swappiness is not 10.  IE just cripple swap all together.
The best is just to enable zram swap so that log2ram and system can make use of swap in memory limited devices such as the pi.
Zram will never be improved it is what it is and its writeback cache could of probably needed improvement but is not necessary and will not be getting improvement.
The code for zram has remained practically static since kernel 3.15 apart from the recent lzo-rle inclusion.

Swappiness has nothing to do with log2ram really unless you are going to take benefit of zram as tmpfs will push out idle pages to zram swap. In the pi the 100mb swap is never used and pretty pointless but it has no function or connection with log2ram.
So Zram can help much with static old logs and the limited space you provide, but hey write what you wish.
If you are using zram swap its actually beneficial to be higher than default as zram is near memcpy speed ram as opposed to assumed HD style media 80 is good approx level.

closed 2019-06-23 by azlux