azlux/log2ram #79: Logs not saved to /var/log/log2ram.log

chris2ao, 2019-05-28:
I've had the log2ram script running for a few days and I'm concerned that the logs are not being written to /var/log/log2ram. I did modify the cronjob save daily instead of hourly. I used the command from the instructions: `sudo mv /etc/cron.hourly/log2ram.hourly /etc/cron.daily/log2ram.daily` but it said "log2ram.hourly" didn't exist, only "log2ram." So I used the mv command and added daily on the end of the file name.

I have verified that the log is created in ram:
pi@raspberrypi:~ $ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        58G  1.4G   54G   3% /
devtmpfs        459M     0  459M   0% /dev
tmpfs           464M  3.4M  460M   1% /dev/shm
tmpfs           464M  6.2M  457M   2% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           464M     0  464M   0% /sys/fs/cgroup
/dev/mmcblk0p1   43M   23M   21M  53% /boot
log2ram         100M   24M   77M  24% /var/log
tmpfs            93M     0   93M   0% /run/user/999
tmpfs            93M     0   93M   0% /run/user/1000

pi@raspberrypi:~ $ mount
/dev/mmcblk0p2 on / type ext4 (rw,noatime)
devtmpfs on /dev type devtmpfs (rw,relatime,size=469540k,nr_inodes=117385,mode=755)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,mode=755)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,release_agent=/lib/systemd/systemd-cgroups-agent,name=systemd)
cgroup on /sys/fs/cgroup/freezer type cgroup (rw,nosuid,nodev,noexec,relatime,freezer)
cgroup on /sys/fs/cgroup/cpu,cpuacct type cgroup (rw,nosuid,nodev,noexec,relatime,cpu,cpuacct)
cgroup on /sys/fs/cgroup/cpuset type cgroup (rw,nosuid,nodev,noexec,relatime,cpuset)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,nosuid,nodev,noexec,relatime,blkio)
cgroup on /sys/fs/cgroup/memory type cgroup (rw,nosuid,nodev,noexec,relatime,memory)
cgroup on /sys/fs/cgroup/devices type cgroup (rw,nosuid,nodev,noexec,relatime,devices)
cgroup on /sys/fs/cgroup/net_cls type cgroup (rw,nosuid,nodev,noexec,relatime,net_cls)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=29,pgrp=1,timeout=0,minproto=5,maxproto=5,direct)
mqueue on /dev/mqueue type mqueue (rw,relatime)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
sunrpc on /run/rpc_pipefs type rpc_pipefs (rw,relatime)
configfs on /sys/kernel/config type configfs (rw,relatime)
/dev/mmcblk0p1 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p2 on /var/hdd.log type ext4 (rw,noatime)
log2ram on /var/log type tmpfs (rw,nosuid,nodev,noexec,relatime,size=102400k,mode=755)
tmpfs on /run/user/999 type tmpfs (rw,nosuid,nodev,relatime,size=94828k,mode=700,uid=999,gid=996)
tmpfs on /run/user/1000 type tmpfs (rw,nosuid,nodev,relatime,size=94828k,mode=700,uid=1000,gid=1000)

However, when I went to the /var/log/ directory to view the log2ram.log file it was empty with no writes. Please see ls -la below:
pi@raspberrypi:/var/log $ ls -la
total 18572
drwxr-xr-x  6 root     root          640 May 29 00:00 .
drwxr-xr-x 13 root     root         4096 May 27 00:14 ..
-rw-r--r--  1 root     root         1380 May 26 00:10 alternatives.log
drwxr-xr-x  2 root     root          100 May 26 02:45 apt
-rw-r-----  1 root     adm        215760 May 29 00:22 auth.log
-rw-r--r--  1 root     root         1374 May 27 11:17 boot.log
-rw-r--r--  1 root     root            0 Apr  8 11:25 bootstrap.log
-rw-------  1 root     utmp         4224 May 29 00:21 btmp
-rw-r-----  1 root     adm        711630 May 29 00:22 daemon.log
-rw-r-----  1 root     adm         16351 May 27 11:37 debug
-rw-r--r--  1 root     root        42536 May 26 02:45 dpkg.log
-rw-r--r--  1 root     root        24000 May 26 00:10 faillog
-rw-r-----  1 root     adm        211637 May 27 11:37 kern.log
-rw-rw-r--  1 root     utmp       292292 May 29 00:22 lastlog
drwxr-x---  2 www-data www-data       80 May 26 00:10 lighttpd
-rw-r--r--  1 root     root            0 May 27 11:17 log2ram.log
-rw-r-----  1 root     adm        206281 May 28 06:25 messages
drwxr-xr-x  2 pihole   pihole         40 May 26 00:10 pihole
-rw-r--r--  1 pihole   pihole          0 May 29 00:00 pihole-FTL.log
-rw-r--r--  1 pihole   pihole        686 May 29 00:00 pihole-FTL.log.1
-rw-r--r--  1 pihole   pihole       2670 May 28 00:00 pihole-FTL.log.2.gz
-rw-r--r--  1 pihole   pihole       3505 May 27 00:00 pihole-FTL.log.3.gz
-rw-r--r--  1 pihole   pihole     737540 May 29 00:27 pihole.log
-rw-r--r--  1 pihole   pihole   13556102 May 29 00:00 pihole.log.1
-rw-r--r--  1 pihole   pihole    1403259 May 28 00:00 pihole.log.2.gz
-rw-r--r--  1 pihole   pihole     803773 May 27 00:00 pihole.log.3.gz
-rw-r--r--  1 root     root         1587 May 26 03:24 pihole_updateGravity.log
drwxr-x---  2 root     adm            40 Jan 31 22:12 samba
-rw-r-----  1 root     adm         63597 May 29 00:22 syslog
-rw-r-----  1 root     adm        944119 May 28 06:25 syslog.1
-rw-r-----  1 root     adm          4704 May 27 11:37 user.log
-rw-rw-r--  1 root     utmp        27648 May 29 00:22 wtmp

Is it supposed to create a new pihole.log everyday or that's not related? Is this how its supposed to work?

Thank you in advance for your help.

Respectfully,

Chris

azlux, 2019-05-28:
Related to #51 and #36

chris2ao, 2019-05-29:
Thank you for the clarification and quick response and I guess everything is working as it should. Is there anyway to check if the cron job is running? I tried `sudo grep CRON /var/log/syslog | grep log2ram` and I did not see the log2ram.daily cron job in the log.

azlux, 2019-05-29:
Elemtent into cron.hourly don't appear on the syslog with the tag `CRON`.
But you have the global call.
`May 29 09:17:01 rasp CRON[8906]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)`

You can also use `sudo grep -i log2ram /var/log/syslog` you will see the systemd and log2ram calls.

chris2ao, 2019-05-29:
I still don't get any records when I used `sudo grep -i log2ram /var/log/syslog`

pi@raspberrypi:~ $ sudo grep -i log2ram /var/log/syslog
pi@raspberrypi:~ $ sudo grep -i log /var/log/syslog
May 29 06:25:02 raspberrypi liblogging-stdlog:  [origin software="rsyslogd" swVersion="8.24.0" x-pid="397" x-info="http://www.rsyslog.com"] rsyslogd was HUPed
pi@raspberrypi:~ $ 
pi@raspberrypi:~ $ 

I mean, ultimately nothing is broke and i can still query the logs. I was just concerned because the readme had the command `sudo mv /etc/cron.hourly/log2ram.hourly /etc/cron.daily/log2ram.daily` but when I tried to run that command it said that log2ram.hourly didn't exist. You may need to update the readme to reflect that.

Thanks again for your help.

74cmonty, 2019-06-23:
Hm... in my case `/var/log/log2ram.log` is empty, too.

What data is supposed to be written in this logfile?
And which event triggers this?

azlux, 2019-06-23:
@74cmonty your answer is here : #36

azlux, 2019-06-23:
@chris2ao can you try switch to rsync ?
On the config : `USE_RSYNC=true`
It should give all action done to syslog.

closed 2020-10-01 by azlux