azlux/log2ram #80: log2ram.service is running, but df -h shows no relevant filesystem log2ram

74cmonty, 2019-06-23:
Hi,
I installed log2ram successfully with install script.

The service is running:
`root@pc6-loxberry:~# systemctl status log2ram
● log2ram.service - Log2Ram
   Loaded: loaded (/etc/systemd/system/log2ram.service; enabled; vendor preset: enabled)
   Active: active (exited) since Sun 2019-06-23 08:38:00 CEST; 15min ago
  Process: 1847 ExecReload=/usr/local/bin/log2ram write (code=exited, status=0/SUCCESS)
  Process: 161 ExecStart=/usr/local/bin/log2ram start (code=exited, status=0/SUCCESS)
 Main PID: 161 (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/log2ram.service

Jun 23 08:50:51 pc6-loxberry systemd[1]: Reloading Log2Ram.
Jun 23 08:50:51 pc6-loxberry systemd[1]: Reloaded Log2Ram.
Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.
`

Checking the mount points chows relevant log2ram entry:
`root@pc6-loxberry:~# mount
/dev/mmcblk0p2 on / type ext4 (rw,noatime)
devtmpfs on /dev type devtmpfs (rw,relatime,size=469544k,nr_inodes=117386,mode=755)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,nosuid,nodev,noexec,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,mode=755)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,release_agent=/lib/systemd/systemd-cgroups-agent,name=systemd)
cgroup on /sys/fs/cgroup/net_cls type cgroup (rw,nosuid,nodev,noexec,relatime,net_cls)
cgroup on /sys/fs/cgroup/devices type cgroup (rw,nosuid,nodev,noexec,relatime,devices)
cgroup on /sys/fs/cgroup/freezer type cgroup (rw,nosuid,nodev,noexec,relatime,freezer)
cgroup on /sys/fs/cgroup/cpuset type cgroup (rw,nosuid,nodev,noexec,relatime,cpuset)
cgroup on /sys/fs/cgroup/cpu,cpuacct type cgroup (rw,nosuid,nodev,noexec,relatime,cpu,cpuacct)
cgroup on /sys/fs/cgroup/memory type cgroup (rw,nosuid,nodev,noexec,relatime,memory)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,nosuid,nodev,noexec,relatime,blkio)
mqueue on /dev/mqueue type mqueue (rw,relatime)
sunrpc on /run/rpc_pipefs type rpc_pipefs (rw,relatime)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=31,pgrp=1,timeout=0,minproto=5,maxproto=5,direct)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
configfs on /sys/kernel/config type configfs (rw,relatime)
/dev/mmcblk0p2 on /var/hdd.log type ext4 (rw,noatime)
log2ram on /var/log type tmpfs (rw,nosuid,nodev,noexec,relatime,size=40960k,mode=755)
tmpfs on /var/log type tmpfs (rw,nosuid,nodev)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev)
tmpfs on /var/tmp type tmpfs (rw,nosuid,nodev)
tmpfs on /opt/loxberry/log/plugins type tmpfs (rw,nosuid,nodev)
tmpfs on /opt/loxberry/log/system_tmpfs type tmpfs (rw,nosuid,nodev)
/dev/mmcblk0p1 on /boot type vfat (rw,noatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/etc/auto.smb on /media/smb type autofs (rw,relatime,fd=6,pgrp=543,timeout=300,minproto=5,maxproto=5,indirect)
tmpfs on /run/user/1001 type tmpfs (rw,nosuid,nodev,relatime,size=94828k,mode=700,uid=1001,gid=1001)
`

However, df -h does not list a relevant filesystem log2ram:
`root@pc6-loxberry:~# df -h
Dateisystem    Größe Benutzt Verf. Verw% Eingehängt auf
/dev/root        30G    4,1G   25G   15% /
devtmpfs        459M       0  459M    0% /dev
tmpfs           464M    720K  463M    1% /dev/shm
tmpfs           464M    6,4M  457M    2% /run
tmpfs           5,0M       0  5,0M    0% /run/lock
tmpfs           464M       0  464M    0% /sys/fs/cgroup
/dev/mmcblk0p1   41M     23M   19M   55% /boot
tmpfs            93M       0   93M    0% /run/user/1001
`

What is the potential root cause for this?

THX

azlux, 2019-06-23:
Hi,
If I read your output from `mount` :
```
[....]
log2ram on /var/log type tmpfs (rw,nosuid,nodev,noexec,relatime,size=40960k,mode=755)
tmpfs on /var/log type tmpfs (rw,nosuid,nodev)
```

`/var/log` is already a tmpfs. So log2ram isn't able to mount his own tmpfs.
Can you check your `/etc/fstab` file ?

Fun fact : log2ram should work. Because the check after the script start will see the a tmpfs mount point.

Az

74cmonty, 2019-06-24:
Your conclusion is absolutely correct; I didn't know that the installed Image (LoxBerry) has implemented some measures to reduce write I/O on disk.

closed 2019-06-24 by 74cmonty