azlux/log2ram #81: Does it really help or is it just "snakeoil"?

systemofapwne, 2019-07-15:
Hello.
I just lately have discovered and installed this script. It made immediately sense to me, that the number of writes to a SD card is very limited and thus should be kept as minimal as possible. But I just noticed maybe one major flaw in this script. **Please correct me, if I'm wrong**. 

For the beginning of my thought-experiment, I neglet wear-levelling to emphasize wear on one single erase-block.

Normally, when a new log entry is getting written to a file, it will be appended to the logfile - Previous erase-blocks will not be touched. If that new log entry falls into a fresh erase-block and also is smaller than the current erase-block (e.g. of size 4 kB), that single erase-block will get written to multiple times, until the next erase-block is reached. This exceptionally hurts, if the length of the new log-entry is small and hurts most, if the entry is of 1 Byte size (=4096 write-cycles to the same block with no wear levelling). But that extreme case is rather rare or non-existant in the wild. This script now tries to avoid multiple-writes to that newly appended erase-block, which in principle is a good practice.
The ony problem I see now is, that with the hourly cronjob, the whole content of a logfile will be written to the SDCard, so in principle many erase-blocks, which's content did not change (e.g. 1 MB = 250 erase-blocks of each 4 kB). And that 24 times a day (equals 6000 writes). These are more write-operations in total than having the worst case of the "1 Byte long log entry"-example above. Especially for big logfiles, the amount of writes amplifies dramatically. You might have saved multiple thousand writes to one erase-block but yet you have performed even more writes in total.

Lets assume that you have wear-levelling available (not every card has this feature - mine does have it). The write operations might get distributed over the whole card, but yet, the whole card will wear out eventually and that even faster, than without this script.

azlux, 2019-07-23:
Hello,
I hope to be able to answer you in suitable English. 

>  That single erase-block will get written to multiple times, until the next erase-block is reached [...] But that extreme case is rather rare or non-existent in the wild.

Not really, in modern SD cards, small numbers of erase blocks are combined into larger units of equal size which are called allocation groups or allocation units or segments. The usual segment size is between 4 MB and 16MB. Mine is 4MB, so the case you explain here append lot of times for small logs.
```
sudo tune2fs -l /dev/mmcblk0p2 | grep Block
Block size:               4096
Blocks per group:         32768
```

> The ony problem I see now is, that with the hourly cronjob, the whole content of a logfile will be written to the SDCard

To avoid that, the `rsync -aXWv` command (and `cp -rfup`) only copy modified content.
After reading (again) the manual, I realize than your reflexion is good, especially for `cp`.

I need to modify my rsync command to `rsync -aXWv --fuzzy --append ` to append changes only, and make it default.

I'm not confident to remove the cron job. Maybe setting the default to daily is a good thing.

Az

linuxtim, 2019-07-23:
If using `--fuzzy`, might be best to use `--delete-after` or `--delete-delay` as well to avoid recopying lograted files which have already been deleted?  Might help in some cases, and probably can't hurt.

Also it would be good to get some sort write stats

tune2fs gives "lifetime writes" data.

Getting raw data from the FTL or flash filesysytem might be better  (to get some idea of write amplification).  mmcutils, smartctl or ubifs might all be useful for this.

systemofapwne, 2019-07-23:
> Hello,
> I hope to be able to answer you in suitable English.
>
>
> Not really, in modern SD cards, small numbers of erase blocks are combined into larger units of equal size which are called allocation [...]

Correct me, if I'm wrong, but according to tune2fs, the block size is 4096 bytes. I do not see how you associate 4 MiB to an 'allocation group' now. Using "Blocks per group" and multiplying it with the block size does end up with 128 MiB, so this won't be the 'allocation group' then, which gets rewritten each time data is appended. 

> To avoid that, the `rsync -aXWv` command (and `cp -rfup`) only copy modified content.
> After reading (again) the manual, I realize than your reflexion is good, especially for `cp`.
>
> I need to modify [...]

Only copying modified data (=append new data of RAM log to existing file on SDCard) might be the  way to go, since this will make sure, that no "already written data" will be written again. Yet, one has to be carefull, whenever a log-rotate etc. has happened.

dset0x, 2019-11-19:
@azlux Hello. I see you have commented above and committed `-W` _and_ `--no-whole-file`. Was that intentional? I take it you meant to replace `--whole-file` with `--append`. Additionally, wouldn't it be a little safer to use `--append-verify`?

azlux, 2019-11-19:
@dset0x I have some race condition issue with `-W`, `--append` and `fussz` #92
I've discuss with some rsync expert, the parameters was making conflics. So I've change the parameters to have the best rysnc use with minimal writes.

closed 2020-01-08 by azlux