azlux/log2ram #84: Use systemd timers instead of cron?

zkemble, 2019-07-26:
Systemd has timers which can be an alternative to cron jobs. Thoughts on moving to timers and no longer needing a dependency on cron?

azlux, 2019-07-27:
For now, Cron is a standard, I don't really know a systemd systel without cron installed.
But you mark a point. Because log2ram is systemd compliant, It's can be a good thing to use systemd for the cron task.

Az

azlux, 2019-12-19:
I just finish my repository for log2ram . https://github.com/azlux/log2ram/commit/13d0058086ff026b2c97723d248532f2237cfa23 -> Better install method

I'm looking for systemd-timer, but I face a problem, I cannot modify the time interval with a variable.

Cron is easy because you can just move the file between `/etc/cron.daily` and `/etc/cron.weekly/`

wizetux, 2020-01-19:
@azlux When you say that you would like to modify the time interval with a variable, how do you mean?   I currently don't see any variable used to determine whether the cron gets placed into the daily or weekly folder during the install process.

azlux, 2020-01-19:
Readme :
```
refresh time:

By default Log2Ram writes to the HardDisk every day. If you think this is too much, you can move /etc/cron.daily/log2ram in an other cron folder, or remove it if you prefer writing logs only at stop/reboot.
```

But if I use systemd-timer, I need:
- [ ] Config file
- [ ] debconf values
- [ ] update postinstall script for `dpkg-reconfigure`

wizetux, 2020-01-20:
Would you like this work in a separate branch in your repository, or rather that I fork the project, and then do a cross fork pull request?

azlux, 2020-01-20:
I think we can work on main branch directly.

azlux, 2020-05-04:
@wizetux can you make a pull request ? I want to check the ways we can implement that.

PR #117 (2020-07-27): [FEATURE] Replaces CRON usages by a proper systemd timer
This patch additionally adds out-of-the-box compatibility for Arch-based distributions.

> Closes #84
> Closes #77

Bye :wave:
commit c399b403 (2020-07-27): Replaces CRON usages by a proper systemd timer

> Closes #84
> Closes #77
closed 2020-07-27 by azlux