azlux/log2ram #89: Raspberry PI 4, 2GB RAM, Arch Linux: ERROR: RAM disk too small. Can't sync.

PieBru, 2019-08-22:
Hi,
I just installed following the standard `curl+tar+chmod+install` sequence on a Raspi4b with an Arch minimal fresh install.

The log2ram service doesn't start, here is the output:
```
$ systemctl status log2ram
* log2ram.service - Log2Ram
   Loaded: loaded (/etc/systemd/system/log2ram.service; enabled; vendor preset: disabled)
   Active: failed (Result: exit-code) since Sat 2019-08-17 14:39:55 UTC; 5 days ago
  Process: 256 ExecStart=/usr/local/bin/log2ram start (code=exited, status=1/FAILURE)
 Main PID: 256 (code=exited, status=1/FAILURE)

Aug 17 14:39:55 k3s-server1 log2ram[256]: ERROR: RAM disk too small. Can't sync.
Aug 17 14:39:55 k3s-server1 log2ram[256]: mail: Cannot start /usr/sbin/sendmail: executable not found (adjust *mta* variable)
Aug 17 14:39:55 k3s-server1 log2ram[256]: /root/dead.letter 8/189
Aug 17 14:39:55 k3s-server1 log2ram[256]: mail: ... message not sent
```

One month ago I successfully installed log2ram on a Raspi3 with Arch, using the exact same personal annotated steps.

Any hint?
Thanks,
Piero

azlux, 2019-08-22:
Your /var/log folder is bigger than the log2ram reserved size.

PieBru, 2019-08-23:
Perfect hit @azlux, solved.

/var/log was over 70MB because of stored journal, even if the Arch OS was just installed the exact same way as the previous Raspi3.
A `journalctl --vacuum-size=1M` was enough to get /var/log into 25M, so `systemctl start log2ram` went OK.

The error message confused me as I was humbling to put also other /var/ subdirs in RAM (i.e. cache, spool), so I understood what was too small, but at a glance I didn't realize what was too large :(

In such cases maybe a message like `ERROR: RAM disk too small, can't sync. Check /var/log size.` may be even more useful to newbies like me.

Thank you.
Piero

closed 2019-08-23 by PieBru