azlux/log2ram #90: ERROR: RAM disk too small. Can't sync.

vsc55, 2019-09-09:
Hi,

It seems to me that the problem has started after upgrading to Raspbian 10, although I'm not sure.

When executing the program it shows this error:
```
# log2ram start
ERROR: RAM disk too small. Can't sync.
```

If we execute `"du -sh"` without the argument `"-t"` show this:
```
# du -sh  /var/hdd.log/
4,0K    /var/hdd.log/
```

But if we execute it with the argument `"-t"`, it does not show any results:
```
# du -sh -t "40M" /var/hdd.log/
```

azlux, 2019-09-09:
Hi,
Can you check the value SIZE in /etc/log2ram.conf
You can compare it with the size of your log with `du -sh /var/log`

vsc55, 2019-09-09:
Hi,

Sorry, my fault 🤦‍♂.
I have in "/var/log" about "80mb".
I have increased SIZE and it has already worked OK.

Thank you very much for the speed 😄

azlux, 2019-09-09:
no prob

mclaser, 2019-10-19:
(coming from thread #58)
du -sh /var/log => 23M
/etc/log2ram.conf/SIZE = 40M

I'm not sure the solution.  Is it to increase SIZE to ?? (2x /var/log size?)
Once the conf is modified do I just 'sudo service log2ram restart' or should I uninstall/reinstall?

azlux, 2019-10-19:
@mclaser should be good. Try to restart and check again. If not working, paste the output of `service log2ram status`

MegaV0lt, 2019-10-20:
Should be safe to set to something like 100M, because tmpfs is only allocating teh space that is realy needed

mclaser, 2019-10-29:
I set SIZE to 80M and I believe all is well now. Thanks for the app and the help!

service log2ram status
● log2ram.service - Log2Ram
   Loaded: loaded (/etc/systemd/system/log2ram.service; enabled; vendor preset: enabled)
   Active: active (exited) since Thu 2019-10-24 11:17:03 PDT; 5 days ago
  Process: 4153 ExecReload=/usr/local/bin/log2ram write (code=exited, status=0/SUCCESS)
  Process: 259 ExecStart=/usr/local/bin/log2ram start (code=exited, status=0/SUCCESS)
 Main PID: 259 (code=exited, status=0/SUCCESS)
    Tasks: 0 (limit: 4915)
   CGroup: /system.slice/log2ram.service

Oct 25 06:25:03 rp2-ph systemd[1]: Reloading Log2Ram.
Oct 25 06:25:03 rp2-ph systemd[1]: Reloaded Log2Ram.
Oct 26 06:25:02 rp2-ph systemd[1]: Reloading Log2Ram.
Oct 26 06:25:02 rp2-ph systemd[1]: Reloaded Log2Ram.
Oct 27 06:25:03 rp2-ph systemd[1]: Reloading Log2Ram.
Oct 27 06:25:03 rp2-ph systemd[1]: Reloaded Log2Ram.
Oct 28 06:25:03 rp2-ph systemd[1]: Reloading Log2Ram.
Oct 28 06:25:03 rp2-ph systemd[1]: Reloaded Log2Ram.
Oct 29 06:25:03 rp2-ph systemd[1]: Reloading Log2Ram.
Oct 29 06:25:03 rp2-ph systemd[1]: Reloaded Log2Ram.

kcalmond, 2020-07-20:
I tried SIZE=100M in /etc/log2ram.conf. 
After reboot I get this error. 

```
pi@cranberry:/var/log $ systemctl status log2ram.service
● log2ram.service - Log2Ram
   Loaded: loaded (/etc/systemd/system/log2ram.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2020-07-20 12:47:27 PDT; 13s ago
  Process: 14389 ExecStart=/usr/local/bin/log2ram start (code=exited, status=1/FAILURE)
 Main PID: 14389 (code=exited, status=1/FAILURE)

Jul 20 12:47:27 cranberry.almond.local systemd[1]: Starting Log2Ram...
Jul 20 12:47:27 cranberry.almond.local log2ram[14389]: ERROR: RAM disk for /var/hdd.log/ too small. Can't sync.
Jul 20 12:47:27 cranberry.almond.local log2ram[14389]: /usr/local/bin/log2ram: 61: /usr/local/bin/log2ram: mail: not found
Jul 20 12:47:27 cranberry.almond.local systemd[1]: log2ram.service: Main process exited, code=exited, status=1/FAILURE
Jul 20 12:47:27 cranberry.almond.local systemd[1]: log2ram.service: Failed with result 'exit-code'.
Jul 20 12:47:27 cranberry.almond.local systemd[1]: Failed to start Log2Ram.
```
Also..
```
pi@cranberry:/var $ df
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       61078228 3379560  55166176   6% /
devtmpfs          823904       0    823904   0% /dev
tmpfs             956000    2452    953548   1% /dev/shm
tmpfs             956000    8676    947324   1% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs             956000       0    956000   0% /sys/fs/cgroup
/dev/mmcblk0p1    258095   55206    202889  22% /boot
tmpfs             191200       0    191200   0% /run/user/999
tmpfs             191200       0    191200   0% /run/user/1000

pi@cranberry:/var $ sudo du -sh /var/log
87M	/var/log
```
Next I tried SIZE=200M in /etc/log2ram.conf. 
That worked. BTW Thanks for this awesome little addition to the pihole world :-).

awiel, 2023-01-10:
For the size of /var/log check journal log: du -sh /var/log/journal

To make the journal log smaller: 
journalctl --rotate
journalctl --vacuum-time=1d

To set a maximum to the journal log:
vi /etc/systemd/journald.conf
SystemMaxUse=50M
and reboot or: systemctl restart systemd-journald

jeffjoker, 2023-11-26:
I think the info : RAM disk for "/var/hdd.log/" is incorrect and could be fixed
I have had this issue when installing, I fixed it by reducing the log sizes (not hdd.log which was an empty folder) byt the mentioned methods (log rotate + max size...)
The issue reappeared recently and I can't seem to fix, even with a 200M size in the log2ram.conf

awiel, 2023-11-26:
Try du command to discover the size of subfolders

Viranchee, 2023-12-13:
I faced this issue and removed journal logs to ~20 MBs and rebooted my machine (RPi Zero W)

But still log2ram exited soon after startup

``` bash
>  [127]> systemctl status log2ram
● log2ram.service - Log2Ram
     Loaded: loaded (/etc/systemd/system/log2ram.service; enabled; vendor preset: enabled)
     Active: active (exited) since Wed 2023-12-13 02:31:49 EST; 6min ago
    Process: 187 ExecStart=/usr/local/bin/log2ram start (code=exited, status=0/SUCCESS)
   Main PID: 187 (code=exited, status=0/SUCCESS)
        CPU: 8.026s

DATE_MACHINE log2ram[230]: journal/0f42d672652246e790e0c71bb5cf550a/user-1000.journal
DATE_MACHINE log2ram[230]: rsync: [receiver] rsync_xal_set: lsetxattr("/var/log/journal/0f42d672652246e790e0c71bb5cf55>
DATE_MACHINE log2ram[230]: private/
DATE_MACHINE log2ram[230]: runit/
DATE_MACHINE log2ram[230]: runit/ssh/
DATE_MACHINE log2ram[230]: rsync: [receiver] rsync_xal_set: lsetxattr("/var/log/journal/0f42d672652246e790e0c71bb5cf55>
DATE_MACHINE log2ram[230]: deleting log2ram.log
DATE_MACHINE log2ram[230]: sent 112,222,475 bytes  received 733 bytes  7,739,531.59 bytes/sec
DATE_MACHINE log2ram[230]: total size is 112,191,727  speedup is 1.00
DATE_MACHINE log2ram[230]: rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.>
```

Any clue how could be the problem or how I can solve this?

beigexperience, 2024-03-27:
Still the same problem on a brand new install of Raspberry Pi os on a Raspi5. Increasing the size and log_disk_size solved the problem.

CordoWEB, 2024-05-06:
The problem is very easy : The initial size of /var/log is bigger than RAM disk. 2 solutions :
 - Reduce /var/log
 - Increase RAM disk