azlux/log2ram #92: Rsync and fuzzy param

TookaFace, 2019-09-18:
Hi, with new version and new param for rsync, i have a problem.

If i create a file, for example, /var/log/foe.log, and reboot, then the file is no longer here with rsync set to true.

If i use cp then file is here at reboot.

I ran some tests, and without fuzzy param, it work, file is here at reboot.

azlux, 2019-09-19:
Hi,
I didn't see this behavior.
Is the file present into /var/hdd.log/foe.log ?

Az

azlux, 2019-09-19:
Ok, I see the issue.
it's appear when you hit race condition.
It's a conflict between `--append` and `--fuzzy`. Append will avoid fuzzy to do is job on smaller file than destination.

TookaFace, 2019-09-19:
Ok, my rsync version is 3.1.2 protocol version 31.

azlux, 2019-09-19:
Can you replace 
`rsync -aXWv --fuzzy --no-whole-file --append --delete-after --links `
by 
`rsync -aXv --inplace --no-whole-file --delete-after `

On the file `/usr/local/bin/log2ram` (2 places on the file. L21 and L41.
It's should be better.

TookaFace, 2019-09-20:
That seems to work, thank you

azlux, 2019-09-20:
New release with the fix. Thank for the report.
https://github.com/azlux/log2ram/releases

closed 2019-09-20 by azlux