azlux/log2ram #96: What if I want another path ?

Vartkat, 2019-12-26:
Hi,

I'm using Home Assistant and would like to include on the RamDisk another path than /var/log. Is that possible ?
Another question would be how to force dump the memory to disk from command line ?

Thanks for your help

Regards
V.

azlux, 2020-01-19:
Hi,
Please upvote this issue if you want this feature.

The following two lines do the path config : https://github.com/azlux/log2ram/blob/master/log2ram#L5-L6

I will Pin this issue and check if people want to have custom folder.

cmonty14, 2020-02-22:
Hi,
I'm facing a very similar issue running [OSMC](https://osmc.tv/) on Raspi 3.
OSMC writes Kodi logfile to _/home/osmc/.kodi/temp/kodi.log_.

Therefore I'm supporting this feature request.

However I want to discuss another workaround and address this question here:
If I create a bind mount `mount --bind /home/osmc/.kodi/temp/ /var/log/kodi`, will this ensure that any write IO goes to ramlog disk only? Or will the write IO go to both directories, _/home/osmc/.kodi/temp/_ and _/var/log/kodi_? In the later case, this is no workaround.

THX

xtvdata, 2020-03-20:
I have a similar need to reduce I/O on cacti rdd files (which is much higher than I/O from logging).

pierrotori, 2020-04-10:
ok thank you but if i add a specifique directory , i duplicate log2ram.conf or i add a variable RAM_LOG2 ?

azlux, 2020-05-03:
this have been commited. You can test it.
I will release this feature after few days without issue.

azlux, 2020-05-20:
After 2 week with no error reported, I think I will release this feature in the next stable version incoming this week.

closed 2020-05-21 by azlux