azlux/log2ram #99: syncFromDisk should use $LOG_DISK_SIZE instead of $SIZE

cotchon, 2020-01-31:
HDD_LOG is compared to $SIZE to check if the logs fits into the newly zram created.
But $SIZE is the mem_limit of zram (physically).
HDD_LOG should be compared to $LOG_DISK_SIZE instead of $SIZE since it is the disk size of the zram. 
proposal :
    if [ ! -z "$(du -sh -t "$SIZE" $HDD_LOG/ | cut -f1)" ]; then
    if [ ! -z "$(du -sh -t "$LOG_DISK_SIZE" $HDD_LOG/ | cut -f1)" ]; then

azlux, 2020-02-05:
You're right, In case of zram, I need to check `$LOG_DISK_SIZE`. I will add another condition.
Thank for the report

azlux, 2020-05-03:
I've some issue on zram for now, are you have to make a pull requests ?

cotchon, 2020-05-05:
Sorry, can you explain what is a pull request please ?

azlux, 2020-05-05:
Github have some explicating into their wiki : https://help.github.com/en/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request-from-a-fork
The goal is to send me the changes you have done into your fork, to be merge into this project.

cotchon, 2020-05-05:
I can not create a branch on the projet, but I replaced in log2ram the following line : 
26:     if [ ! -z "$(du -sh -t "$SIZE" $HDD_LOG/ | cut -f1)" ]; then
by:
26:    if [ ! -z "$(du -sh -t "$LOG_DISK_SIZE" $HDD_LOG/ | cut -f1)" ]; then

azlux, 2020-05-06:
ok, I will do that.
Thank

azlux, 2020-05-07:
Should be fixed on https://github.com/azlux/log2ram/commit/881eea145f8393c17cb0b18f4b6113cf1a1cc96d

commit 881eea14 (2020-05-07): add size check for zram

fix #99
closed 2020-05-07 by azlux