azlux/log2ram #100: fd report 100% usage on /var/log

ChipTechno, 2020-02-04:
Hello. I want to ask about a strange behavior of log2ram. 
I have log2ram running on 3 Raspberry3b. 
with <code>df</code> command, the system report that <code>/var/log</code> is 100% usage and 0% free space. But when i check the <code>/var/log</code> directory, there are only some MB of logs in it. Restart the log2ram systemd service restore the usage of <code>/var/log</code> to 0%.

### Problem
- df report 100% disk usage when only few MB are used.

### log2ram config
<pre>

# Configuration file for Log2Ram (https://github.com/azlux/log2ram) under MIT license.
# This configuration file is read by the log2ram service

# Size for the ram folder, it defines the size the log folder will reserve into the RAM.
# If it's not enough, log2ram will not be able to use ram. Check you /var/log size folder.
# The default is 40M and is basically enough for a lot of applications.
# You will need to increase it if you have a server and a lot of log for example.
SIZE=40M

# This variable can be set to true if you prefer "rsync" rather than "cp".
# I use the command cp -u and rsync -X, so I don't copy the all folder every time for optimization.
# You can choose which one you want. Be sure rsync is installed if you use it.
USE_RSYNC=false

# If there are some errors with available RAM space, a system mail will be send
# Change it to false and you will have only a log if there is no place on RAM anymore.
MAIL=true

# **************** Zram backing conf  *************************************************

# ZL2R Zram Log 2 Ram enables a zram drive when ZL2R=true ZL2R=false is mem only tmpfs
ZL2R=false
# COMP_ALG this is any compression algorithm listed in /proc/crypto
# lz4 is fastest with lightest load but deflate (zlib) and Zstandard (zstd) give far better compression ratios
# lzo is very close to lz4 and may with some binaries have better optimisation
# COMP_ALG=lz4 for speed or Zstd for compression, lzo or zlib if optimisation or availabilty is a problem
COMP_ALG=lz4
# LOG_DISK_SIZE is the uncompressed disk size. Note zram uses about 0.1% of the size of the disk when not in use
# LOG_DISK_SIZE is expected compression ratio of alg chosen multiplied by log SIZE
# lzo/lz4=2.1:1 compression ratio zlib=2.7:1 zstandard=2.9:1
# Really a guestimate of a bit bigger than compression ratio whilst minimising 0.1% mem usage of disk size
LOG_DISK_SIZE=100M
</pre>

### Other logs
<pre>
pi@RevPi18034:~ $ df
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root        3643096 2713564    738060  79% /
devtmpfs          469908       0    469908   0% /dev
tmpfs             474516       0    474516   0% /dev/shm
tmpfs             474516   47848    426668  11% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs             474516       0    474516   0% /sys/fs/cgroup
/dev/mmcblk0p1     44220   21834     22387  50% /boot
log2ram            40960   40960         0 100% /var/log
tmpfs              94900       0     94900   0% /run/user/1000
pi@RevPi18034:~ $ ls /var/log
apache2  log2ram.log
pi@RevPi18034:~ $ ls -la --block-size=M /var/log
total 1M
drwxr-xr-x  3 root root 1M Jan 17 06:25 .
drwxr-xr-x 13 root root 1M Nov  4  2016 ..
drwxr-xr-x  2 root adm  1M Jan 16 14:46 apache2
-rw-r--r--  1 root root 0M Jan 17 06:25 log2ram.log
pi@RevPi18034:~ $ sudo- systemctl status log2ram
-bash: sudo-: command not found
pi@RevPi18034:~ $ sudo systemctl status log2ram
● log2ram.service - Log2Ram
   Loaded: loaded (/etc/systemd/system/log2ram.service; enabled; vendor preset: enabled)
   Active: active (exited) since Sun 2019-12-15 13:38:47 JST; 1 months 20 days ago
  Process: 2776 ExecReload=/usr/local/bin/log2ram write (code=exited, status=0/SUCCESS)
  Process: 831 ExecStart=/usr/local/bin/log2ram start (code=exited, status=0/SUCCESS)
 Main PID: 831 (code=exited, status=0/SUCCESS)
    Tasks: 0 (limit: 4915)
   CGroup: /system.slice/log2ram.service

Feb 04 06:25:02 RevPi18034 systemd[1]: Reloading Log2Ram.
Feb 04 06:25:02 RevPi18034 systemd[1]: Reloaded Log2Ram.
Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.
pi@RevPi18034:~ $ sudo systemctl restart log2ram
pi@RevPi18034:~ $ df
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root        3643096 2713564    738060  79% /
devtmpfs          469908       0    469908   0% /dev
tmpfs             474516       0    474516   0% /dev/shm
tmpfs             474516   47848    426668  11% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs             474516       0    474516   0% /sys/fs/cgroup
/dev/mmcblk0p1     44220   21834     22387  50% /boot
tmpfs              94900       0     94900   0% /run/user/1000
log2ram            40960       0     40960   0% /var/log
pi@RevPi18034:~ $ sudo systemctl status log2ram
● log2ram.service - Log2Ram
   Loaded: loaded (/etc/systemd/system/log2ram.service; enabled; vendor preset: enabled)
   Active: active (exited) since Tue 2020-02-04 08:56:55 JST; 19s ago
  Process: 13044 ExecStop=/usr/local/bin/log2ram stop (code=exited, status=0/SUCCESS)
  Process: 2776 ExecReload=/usr/local/bin/log2ram write (code=exited, status=0/SUCCESS)
  Process: 13051 ExecStart=/usr/local/bin/log2ram start (code=exited, status=0/SUCCESS)
 Main PID: 13051 (code=exited, status=0/SUCCESS)

Feb 04 08:56:55 RevPi18034 systemd[1]: Starting Log2Ram...
Feb 04 08:56:55 RevPi18034 systemd[1]: Started Log2Ram.
pi@RevPi18034:~ $ ls -la --block-size=M /var/log
total 1M
drwxr-xr-x  3 root root 1M Jan 17 06:25 .
drwxr-xr-x 13 root root 1M Nov  4  2016 ..
drwxr-xr-x  2 root adm  1M Jan 16 14:46 apache2
-rw-r--r--  1 root root 0M Feb  4 08:56 log2ram.log
</pre>

azlux, 2020-02-05:
Hi,
I read you debug and I see some mistakes into you search.
The command `dh` show you the size of the folder `/var/log` because it's a mount point but `ls -alih` don't.
You can see a folder `apache` with 1M size, this don't give you the size of the content of this folder.
You need to use `du -sh /var/log/*` to see real size. I personally use the tool `ncdu`.
I suppose apache take more than 40Mo depending of you traffic and logrotate you have.
Azlux

ChipTechno, 2020-02-05:
Hi @azlux 
I found the problem. I'm using the RP3 Compute module with only 4GB of eMMC (500MB free after flash the raspbian image). The kernel log(kern.log) is quite big, ~5MB a day that flood the /var/log partition.
Thank for your support!

closed 2020-02-05 by ChipTechno