azlux/log2ram #101: I've updated but still have hourly update

Barabba11, 2020-03-11:
Hi! I've update to change from hourly to daily, but it still write all files every hour, as in syslog:
Mar 11 18:17:01 raspberrypi CRON[20554]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Mar 11 18:17:01 raspberrypi systemd[1]: Reloading Log2Ram.
Mar 11 18:17:01 raspberrypi systemd[1]: Reloaded Log2Ram
Mar 11 19:17:01 raspberrypi CRON[25188]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Mar 11 19:17:01 raspberrypi systemd[1]: Reloading Log2Ram.
Mar 11 19:17:01 raspberrypi systemd[1]: Reloaded Log2Ram.

I've follower the instruction, stopped service before, update database and program, all seems was ok. WHere is the problem? Old settings keep there?

azlux, 2020-03-11:
hi,
which method do you used ?

If you see log2ram into `/etc/cron.daily/log2ram` and `/etc/cron.hourly/log2ram` you can remove the first one.

Barabba11, 2020-03-11:
thanks for fast answer, I had them on both, I've removed the one on the hourly folder

closed 2020-03-11 by azlux