azlux/log2ram #106: Avoid logging into syslog (question)

k-sartor, 2020-05-06:
Hi,
i would like to know if there is a way to turn off loging of log2ram into syslog file while it has its own log file.

I tried by adding a rule in rsyslog.conf
:msg, contains, "log2ram" stop
but it does not work.
Thank you

SargonofAssyria, 2020-05-09:
This works for me:
`
if $programname == 'log2ram' then /dev/null
& stop
`

k-sartor, 2020-05-09:
Thank you. 
However, in this case, it is required to see in syslog when log2ram have been restarted (by systemd) and in the log2ram.log the files synced.
Is there any way to put the date of file syning in log2ram.log?

azlux, 2020-07-22:
@k-sartor new changes will use systemd-timer. So you will have everything into systemd logs instead of syslog.

azlux, 2020-10-01:
A new version without cron have been release. This issue should be fix because I now use systemd-timer

closed 2020-10-12 by azlux