azlux/log2ram #108: preserve option not working on Raspbian Buster

SargonofAssyria, 2020-05-09:
I get this error "cp: cannot preserve security context without an SELinux-enabled kernel"
So the copy function did not work.
My system:
RPi 4B with Linux domoticz 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux

Could the type of kernel be checked and then copy depending on the type?

azlux, 2020-05-09:
Is that a warning or an error ?
I suppose you can use rsync, it is able to make the good copy independing of the SELinux enable or not.

closed 2020-05-09 by azlux

SargonofAssyria, 2020-05-09:
The problem is that I had to make a new fresh install.
Then the problem came that I had almost an empty /var/log directory.
Because of that mosquitto did not start and it did function normally before the reboot.
I found the problem, but it might be an idea to prevent others from a search to this kind of problem.
It is just a suggestion to help others who might have the same trouble.
I solved mine.

commit 3efc8f35 (2020-05-09): preserve context make issue on many OS

fix #108