azlux/log2ram #110: log2ram timed out at boot

MegaV0lt, 2020-05-14:
On Debian 10.4 i get
```
[    5.245706] sd 2:0:0:0: [sdb] Attached SCSI disk
[    5.800987] EXT4-fs (sdb1): mounted filesystem with ordered data mode. Opts: (null)
[    7.614944] r8169 0000:02:00.0 enp2s0: Link is Up - 1Gbps/Full - flow control rx/tx
[    7.614957] IPv6: ADDRCONF(NETDEV_CHANGE): enp2s0: link becomes ready
[    9.669877] systemd[1]: systemd-rfkill.service: Succeeded.
[  124.084928] systemd[1]: log2ram.service: Start operation timed out. Terminating.
[  124.085428] systemd[1]: log2ram.service: Main process exited, code=killed, status=15/TERM
[  124.085556] systemd[1]: log2ram.service: Failed with result 'timeout'.
[  124.086432] systemd[1]: Failed to start Log2Ram.
[  124.089565] systemd[1]: Starting Journal Service...
[  124.250651] systemd[1]: Started Journal Service.
[  124.262447] systemd-journald[553]: Received request to flush runtime journal from PID 1
[  124.957245] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[  124.957247] Bluetooth: BNEP filters: protocol multicast
[  124.957251] Bluetooth: BNEP socket layer initialized
```
Delays the boot for 2 Minutes

Cron also fails:
```
/etc/cron.daily/log2ram:
log2ram.service is not active, cannot reload.
run-parts: /etc/cron.daily/log2ram exited with return code 1
```
mount
```
[~] # mount |grep log
/dev/sda2 on /var/hdd.log type ext4 (rw,noatime,nodiratime,errors=remount-ro,stripe=32696)
/dev/sda2 on /mnt/root_fs/var/hdd.log type ext4 (rw,noatime,nodiratime,errors=remount-ro,stripe=32696)
log2ram on /var/log type tmpfs (rw,nosuid,nodev,noexec,relatime,size=153600k,mode=755)
log2ram on /mnt/root_fs/var/log type tmpfs (rw,nosuid,nodev,noexec,relatime,size=153600k,mode=755)
log2ram on /mnt/root_fs/var/hdd.log type tmpfs (rw,nosuid,nodev,noexec,relatime,size=153600k,mode=755)

```

azlux, 2020-05-14:
It's come from an OS upgrade ? 
Is there are /var/log/log2ram.log logs ?

MegaV0lt, 2020-05-14:
log2ram.log is empty
Have no idea. what happened. 
What could be the reason for timeot at boot? Can you add some debug output?

azlux, 2020-05-14:
is `journalctl -t log2ram -n 50` give you something ?

MegaV0lt, 2020-05-14:
Nothing after Mai-09
```
Mai 09 18:06:35 IdeaCentre-200 log2ram[305]: samba/log.2003_ec_5bed_df00_222_4dff_fe84_552d
Mai 09 18:06:35 IdeaCentre-200 log2ram[305]: samba/log.2003_ec_5bee_5d00_222_4dff_fe84_552d
Mai 09 18:06:35 IdeaCentre-200 log2ram[305]: samba/log.2003_ec_5bee_8900_222_4dff_fe84_552d
Mai 09 18:06:35 IdeaCentre-200 log2ram[305]: samba/log.2003_ec_5bee_d100_222_4dff_fe84_552d
Mai 09 18:06:35 IdeaCentre-200 log2ram[305]: samba/log.2003_ec_5bf1_4200_222_4dff_fe84_552d
Mai 09 18:06:35 IdeaCentre-200 log2ram[305]: samba/log.2003_ec_5bf1_d400_222_4dff_fe84_552d
Mai 09 18:06:35 IdeaCentre-200 log2ram[305]: samba/log.2003_ec_5bf1_ee00_222_4dff_fe84_552d
Mai 09 18:06:35 IdeaCentre-200 log2ram[305]: samba/log.2003_ec_5bf2_8900_222_4dff_fe84_552d
Mai 09 18:06:35 IdeaCentre-200 log2ram[305]: samba/log.2003_ec_5bf2_9b00_222_4dff_fe84_552d
lines 20-51/51 (END)
```

MegaV0lt, 2020-05-14:
Service 
```
root auf IdeaCentre-200 am 14.05.2020 11:20 
[~] # service log2ram status
● log2ram.service - Log2Ram
   Loaded: loaded (/etc/systemd/system/log2ram.service; enabled; vendor preset: enabled)
   Active: failed (Result: timeout) since Sat 2020-05-09 18:06:35 CEST; 4 days ago
 Main PID: 305 (code=killed, signal=TERM)

Mai 09 18:06:35 IdeaCentre-200 log2ram[305]: samba/log.2003_ec_5bf1_4200_222_4dff_fe84_552d
Mai 09 18:06:35 IdeaCentre-200 log2ram[305]: samba/log.2003_ec_5bf1_d400_222_4dff_fe84_552d
Mai 09 18:06:35 IdeaCentre-200 log2ram[305]: samba/log.2003_ec_5bf1_ee00_222_4dff_fe84_552d
Mai 09 18:06:35 IdeaCentre-200 log2ram[305]: samba/log.2003_ec_5bf2_8900_222_4dff_fe84_552d
Mai 09 18:06:35 IdeaCentre-200 log2ram[305]: samba/log.2003_ec_5bf2_9b00_222_4dff_fe84_552d
Mai 10 07:38:35 IdeaCentre-200 systemd[1]: log2ram.service: Unit cannot be reloaded because it is inactive.
Mai 11 07:38:14 IdeaCentre-200 systemd[1]: log2ram.service: Unit cannot be reloaded because it is inactive.
Mai 12 07:37:13 IdeaCentre-200 systemd[1]: log2ram.service: Unit cannot be reloaded because it is inactive.
Mai 13 07:36:02 IdeaCentre-200 systemd[1]: log2ram.service: Unit cannot be reloaded because it is inactive.
Mai 14 07:39:15 IdeaCentre-200 systemd[1]: log2ram.service: Unit cannot be reloaded because it is inactive.
```

azlux, 2020-05-14:
You have a lot of mount point. That is weird.
I need some time to write a debug version.
Maybe you can test line by line the `start) ` function.

MegaV0lt, 2020-05-14:
i will try to update to the last verison and uninstall and reinstall...

MegaV0lt, 2020-05-14:
Maybe the "wait_for" is waiting too long. What you think about adding a timeot there?

Sample:
```
wait_for () {
    while ! grep -qs "$1" /proc/mounts ; do
        sleep 0.5 ; i=$((i + 1))
        if [ "$i" -gt 180 ] ; then  # Max. 90 seconds
            echo "Timeout waiting for $1"
            exit 1
        fi
    done
}
```

MegaV0lt, 2020-05-14:
reinstalles and reooted. Now it looks like is running without issue; strange...

```
root auf IdeaCentre-200 am 14.05.2020 12:35 
[~] # service log2ram status
● log2ram.service - Log2Ram
   Loaded: loaded (/etc/systemd/system/log2ram.service; enabled; vendor preset: 
   Active: active (exited) since Thu 2020-05-14 12:34:56 CEST; 1min 27s ago
  Process: 319 ExecStart=/usr/local/bin/log2ram start (code=exited, status=0/SUC
 Main PID: 319 (code=exited, status=0/SUCCESS)

Mai 14 12:34:57 IdeaCentre-200 log2ram[319]: building file list ... done
Mai 14 12:34:57 IdeaCentre-200 log2ram[319]: ./
Mai 14 12:34:57 IdeaCentre-200 log2ram[319]: deleting log2ram.log
Mai 14 12:34:57 IdeaCentre-200 log2ram[319]: sent 49 bytes  received 18 bytes  1
Mai 14 12:34:57 IdeaCentre-200 log2ram[319]: total size is 0  speedup is 0.00
Warning: Journal has been rotated since unit was started. Log output is incomple
lines 1-12/12 (END)
```

```
[~] # mount |grep log
/dev/sda2 on /var/hdd.log type ext4 (rw,noatime,nodiratime,errors=remount-ro,stripe=32696)
/dev/sda2 on /mnt/root_fs/var/hdd.log type ext4 (rw,noatime,nodiratime,errors=remount-ro,stripe=32696)
log2ram on /var/log type tmpfs (rw,nosuid,nodev,noexec,relatime,size=102400k,mode=755)
log2ram on /mnt/root_fs/var/log type tmpfs (rw,nosuid,nodev,noexec,relatime,size=102400k,mode=755)
log2ram on /mnt/root_fs/var/hdd.log type tmpfs (rw,nosuid,nodev,noexec,relatime,size=102400k,mode=755)

```

MegaV0lt, 2020-05-14:
A lot of dirs have gone!

```
root auf IdeaCentre-200 am 14.05.2020 12:39 
[~] # ll /var/log
insgesamt 376K
-rw-r----- 1 root   adm     1,8K Mai 14 12:40 auth.log
-rw-rw---- 1 root   utmp       0 Mai 14 12:34 btmp
-rw-r----- 1 root   adm      44K Mai 14 12:39 daemon.log
-rw-r----- 1 root   adm      13K Mai 14 12:35 debug
-rw------- 1 root   root    2,2K Mai 14 12:35 fail2ban.log
-rw-r----- 1 root   adm      81K Mai 14 12:35 kern.log
-rw-rw-r-- 1 root   utmp     292 Mai 14 12:35 lastlog
-rw-r----- 1 root   adm      69K Mai 14 12:35 messages
-rw-r--r-- 1 root   root    1,1K Mai 14 12:36 monitorix
-rw------- 1 nobody nogroup  480 Mai 14 12:41 monitorix-httpd
-rw------- 1 root   root     185 Mai 14 12:35 php7.3-fpm.log
drwx------ 2 root   root      40 Mai 14 12:34 private
-rw-r----- 1 root   adm     127K Mai 14 12:40 syslog
drwxr-xr-x 2 root   root      60 Mai 14 12:35 unattended-upgrades
-rw-r----- 1 root   adm      126 Mai 14 12:34 user.log
-rw-rw-r-- 1 root   utmp    1,9K Mai 14 12:35 wtmp
root auf IdeaCentre-200 am 14.05.2020 12:41 
[~] # ll /var/hdd.log/
insgesamt 0
```

azlux, 2020-05-14:
Are you using the git or apt ?

MegaV0lt, 2020-05-14:
Latest git

azlux, 2020-05-14:
the config have change since last commit, maybe some conflicts ?

MegaV0lt, 2020-05-14:
No, just setting to 100M and to use rsync.

azlux, 2020-05-17:
I'm blocking with your issue, I've make lot of change and I want to release a new version. But since I don't know why It's have failed on yours, I cannot.

MegaV0lt, 2020-05-18:
Just make the new version, i will check then again. dont worry ;-)

azlux, 2020-07-22:
Can I close this issue ? 2 version have been release since this ticket is open.

MegaV0lt, 2020-07-23:
Well, can be closed for now. I report when the issue comes up again.

closed 2020-07-23 by MegaV0lt