azlux/log2ram #113: Updating to 1.5.1 breaks system and makes files read-only

DIGIN4, 2020-05-21:
After upgrading to 1.5.1 from 1.4.4 both of my Raspbian systems got broken, and it seems files became read-only after rebooting, I checked /etc/log2ram.conf but didn't change anything, I noticed this in the release info: "Add an additionnal path (in the config) - BREAKING, PLEASE USE THE NEW CONFIG" but instructions & comments aren't clear.

azlux, 2020-05-21:
Hi,
The change is this line : https://github.com/azlux/log2ram/blob/master/log2ram.conf#L21
I've add an additionnal important config, BUT version 1.5.1 fix the case this config isn't exist.

You said you have read-only files, can you explain ? Which ones ?

azlux, 2020-05-21:
This line make readonly mount point ONLY if it's not able to write into the HDD folder : https://github.com/azlux/log2ram/blob/master/log2ram#L97

DIGIN4, 2020-05-21:
Files in /etc/ such as apt files and even log2ram.conf became read-only

azlux, 2020-05-21:
What the output of `mount` and `df -h`?

RapFan89, 2020-05-22:
i have the same problem, what can we do?

azlux, 2020-05-22:
` /dev/mmcblk0p2 on / type ext4 (ro,noatime)` I see the issue.
Can I have the config file ?

azlux, 2020-05-22:
I've rollback the affecting code into 1.5.2

Grabou, 2020-05-22:
Same here. Reinstalling my Raspbian (only for Pi-hole x log2ram) cause I'm a Linux noob. :D

DIGIN4, 2020-05-22:
> Same here. Reinstalling my Raspbian (only for Pi-hole x log2ram) cause I'm a Linux noob. :D

You should create backup images in case something breaks

azlux, 2020-05-22:
@Grabou I understand, My software should break anything. Did you update to last version just released ?

Grabou, 2020-05-22:
> You should create backup images in case something breaks

Yeah. I need 15 min to reinstall properly my RPi for Pi-hole. That's not a big issue for me. I should follow your advice next time.

Grabou, 2020-05-22:
> 
> 
> @Grabou I understand, My software should break anything. Did you update to last version just released ?

I updated yesterday to 1.5.1 and wanted to reupdate today as new version appeared (GitHub release notification). And "oops, read only everywhere". :D

RapFan89, 2020-05-22:
to remount the os in rw simply type

sudo mount -o rw,remount /dev/mmcblk0p2 /

so you don't need to reinstall anything

Grabou, 2020-05-22:
> to remount the os in rw simply type
> 
> sudo mount -o rw,remount /dev/mmcblk0p2 /
> 
> so you don't need to reinstall anything

Too late, but good to know. May be useful for others. Thanks!

DIGIN4, 2020-05-22:
I have upgraded from 1.4.4 to 1.5.2 and it's working fine.

closed 2020-05-22 by azlux