azlux/log2ram #114: log2ram removed folders under /var/log

JonRobertson, 2020-05-22:
log2ram removed several folders under /var/log, including the apache2 folder. Now apache2 refuses to start.

azlux, 2020-05-22:
maybe related to #113 ?

JonRobertson, 2020-05-22:
I don't know. My file system did become read-only, so I suspect that is related. But I don't see anything in #113 related to folders being removed from /var/log. I had /var/log in PATH_DISK. Based on the core purpose of log2ram, I would expect it to handle all subfolders under /var/log without me doing anything to the config file.

I've uninstalled log2ram. I just wanted to let you know about the issue.

MegaV0lt, 2020-05-22:
Had the same issue. after uninstalling and reinstalling most of subdirs in /var/log were missing...

azlux, 2020-05-26:
Is this issue related to version `1.5` or should I search the issue on current version ?

MegaV0lt, 2020-05-26:
No idea what happened in my case. I was updating installation that was timed out at boot...

JonRobertson, 2020-05-26:
I uninstalled log2ram and have not installed it since you released 1.5. I don't mind trying it again. However, I'm relatively new to running a Linux server and it takes me a bit to find a way to fix things when they break or don't work.

Is there anything I need to change in the default config file if all I want is for /var/log (and all subfolders) to be written to ram?

A comment in the config file states "the `/path/hdd.folder` will be automatically created". What does this mean? What is /path/hdd.folder?

azlux, 2020-05-26:
@JonRobertson this path is to store /var/log into the HDD. You can remove it if it existe.

JonRobertson, 2020-05-26:
My apologies, please bare with me. Do you mean remove the folder or remove the PATH_DISK setting from the config file? I would think the code that sets PATH_DISK would be needed.

If I leave log2ram.conf alone:
```
# Variable for folders to put in RAM. You need to specify the real folder `/path/folder` , the `/path/hdd.folder` will be automatically created. Multiple path can be separeted by `;`. Do not add the final `/` !
PATH_DISK="/var/log"
```
Does var2log create a folder with the path /var/hdd.log ?

azlux, 2020-07-22:
@JonRobertson you're right, log2ram create the needed folders.
I will close this issue, since it's related to a rollbacked code.

closed 2020-07-22 by azlux