azlux/log2ram #115: No cron job?

bondskin, 2020-06-05:
Hi,

I cannot find any cron job to write the content to SD card once per day, as mentioned in the readme.
Where would I find this job or how can I run this manually?

azlux, 2020-07-22:
Can you try to use rsync by default ?

azlux, 2020-10-01:
A new version have been release. Can you test again with it ?

ZeroFux78, 2020-10-29:
I don't think the logs have been rotating since the cron was removed.  I reinstalled log2ram this evening (Log2Ram was at 100% of 40M, so I increased it to 64M) and set rsync to true in the conf file. I've reloaded the timer and it doesn't look like the log has moved/rotated/cleared to the SD.

azlux, 2020-10-29:
Since the upgrade, the flush is done once a day.
You can check with `sudo journalctl -t log2ram` the log2ram activities.

ZeroFux78, 2020-10-30:
Thank you for getting back to me!  Here's my df -h:
`log2ram         128M  114M   15M  89% /var/log`

Here is my result from the journalctl command you provided:
```
$ sudo journalctl -t log2ram
-- Logs begin at Thu 2020-10-29 15:32:26 PDT, end at Thu 2020-10-29 22:54:33 PDT.
-- No entries --
```

I had an issue with my pihole earlier today and I had to restart my pi this afternoon.  I'll check the journal tomorrow morning and see if the log rotated prior to midnight.

ZeroFux78, 2020-11-08:
Sorry about the delay in responding.  I dropped the ball on this. I checked my journal and process did run, but the log is still capping out.  Everytime I increase the size, 40M to 64M to 85M to 100M, the log just reaches 100% in a day or two. 

I thought log2ram saved writes to the SD card by writing the log to the ram partition, then nightly writes the log to disk. Once the log is written to disk, log2ram should reset back to 0M for the next days logs, then writes those logs to the disk the next night.   

Is that not the expected process?   Am I supposed to trim my logs manually once they get to large?

xtvdata, 2020-11-08:
> I checked my journal and process did run, but the log is still capping out. Everytime I increase the size, 40M to 64M to 85M to 100M, the log just reaches 100% in a day or two.
> 

This is probably due to the fact that either:
- your log rotation is not working (i.e. you installed something manually that creates significant log that are not rotated and compressed),
- you have issues somewhere in you your system generating too much logging:
  - you have something installed that is periodically generating a lot of errors in the logs,
  - you can have something installed that is logging with "Debug" detail level (generating huge logs).

Thus, I don't think that your issue is linked to _log2ram_, but to something else on your system.

I'd suggest you to read your logs and see what is generating so much volume.

Just to give you an example: I just checked my _/var/log_ dir (that in my case includes inbound rsyslog from 7 external systems, thus potentially 7 times larger than needed normally) and size is below 70MB. How to get there? You need to manually tune the log details -rsyslog- and log rotation -logrotate-, not _log2ram_.
_log2ram_ only required setup is to accomodate enough space to keep the full _/var/log_ in RAM. As best practice I'd suggest to set _log2ram_ to at least twice as much as your "stable" _/var/log_ size (at least twice because issues in the system can happen, and logs will grow, but that is the reason why logs are produced... ;-) ).
Anyway, again, if /var/log is too big you need to work somewhere else to reduce its size. RPI is small, not expensive and a fantastic device, but has limited resources and you need to use them well.

A good practice is to monitor you logs. There are various tools you can use to monitor logs, but the easiest and most lightweight is probably _logwatch_.

> I thought log2ram saved writes to the SD card by writing the log to the ram partition, then nightly writes the log to disk. Once the log is written to disk, log2ram should reset back to 0M for the next days logs, then writes those logs to the disk the next night.

It simply doesn't work like that, the RAM disk is __not__ _flushed_ to disk, it's _synched_ to disk; meaning that the content of the RAM disk is _copied_ to the SD card for persistence, not _moved_.
In every UNIX system the limitation of log size is achieved via the log rotation process.

> Is that not the expected process?

Yes, _log2ram_ leverages _differential_ synching capabilities of Rsync, but that means only to limit the disk access to what is strictly required, updating only files with changes.

What you are thinking about is a different mechanism that is called "overlay", which is usually used to keep the main filesystem in read-only and writing differences to an "overlaid" partition. However this implies continuous access to the base filesystem to read the base data, to the RAM disk to read the diffs, apply the diffs in ram to the base data and update the diffs at each write operation.
This is not in scope for _log2ram_ since the target here is to limit as much as possible access to the SD card (in write to save the physical SD card AND in read to have IO free for the system - as said RPI have very limited IO capabilities and it's much better to avoid using IO since in my experience is the most scarce resource in the system).

> Am I supposed to trim my logs manually once they get to large?

Yes, but not manually. You should probably configure logrotate or check of issues (see above).

---

References:
- rsync - https://en.wikipedia.org/wiki/Rsync
- rsyslog - https://www.rsyslog.com/doc/v8-stable/configuration/index.html
- logorate - https://manpages.debian.org/jessie/logrotate/logrotate.8.en.html
- logwatch - https://www.digitalocean.com/community/tutorials/how-to-install-and-use-logwatch-log-analyzer-and-reporter-on-a-vps

P.S.: sorry for the length, since I noticed most of the issues opened are related to the same point, I hope this can explain a bit more the mechanics in _log2ram_ (which is pretty simple - as it should be).

closed 2022-03-04 by azlux

luigir-it, 2024-07-05:
> However this implies continuous access to the base filesystem to read the base data, to the RAM disk to read the diffs, apply the diffs in ram to the base data and update the diffs at each write operation.

I don't think this statement is entirely accurate.

OverlayFS does make a full copy of the new/edited file in the upper layer. Any subsequent access to that file (whether read or write) [is done against the upper layer](https://docs.kernel.org/filesystems/overlayfs.html#non-directories), which in this case is the RAM.
This means that write operations do not wear the disk at all (unless when you sync back the changes, which is something OverlayFS can't do, you have to implement it yourself) 

"Continuous access" to the base filesystem happens mainly when you need to read files that haven't been modified (and therefore still reside in the lower layer) multiple times. However, reading files does not wear the disk as much as writing does. Besides, once you read data from a file, it usually stays in RAM until the files is closed, reducing repeated reads from the disk.

Moreover, unless you always read all the logs, OverlayFS may actually reduce total reads, since you don't need to copy all the content from disk to RAM every time the service starts.

Finally, this saves some (sometimes a lot) of RAM.

> This is not in scope for log2ram
That said, I think it is...
The only downside I can think of is spreading read calls throughout the life of the service (instead of doing them all at once at the beginning), which could cause (minimal) IO delay.