azlux/log2ram #118: RAM Disk too small error

leonbrag, 2020-07-17:
This is VM with limited memory:

```
leonbrag@minivm:/var/log$ free -h
              total        used        free      shared  buff/cache   available
Mem:          2.7Gi       1.2Gi       322Mi        52Mi       1.3Gi       1.5Gi
Swap:         2.0Gi          0B       2.0Gi
```

Disks:

```
leonbrag@minivm:/var/log$ df -h
Filesystem      Size  Used Avail Use% Mounted on
udev            920M     0  920M   0% /dev
tmpfs           193M  1.8M  191M   1% /run
/dev/sda2       125G   13G  106G  11% /
tmpfs           962M  5.3M  957M   1% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           962M     0  962M   0% /sys/fs/cgroup
/dev/loop1      9.2M  9.2M     0 100% /snap/canonical-livepatch/95
/dev/loop0       55M   55M     0 100% /snap/core18/1754
/dev/loop3       97M   97M     0 100% /snap/core/9436
/dev/loop4       55M   55M     0 100% /snap/core18/1880
/dev/loop2       97M   97M     0 100% /snap/core/9665
/dev/loop5      5.9M  5.9M     0 100% /snap/htop/1222
/dev/loop6      5.9M  5.9M     0 100% /snap/htop/1207
/dev/loop10      30M   30M     0 100% /snap/snapd/8140
/dev/loop7       72M   72M     0 100% /snap/lxd/16044
/dev/loop8       72M   72M     0 100% /snap/lxd/16100
/dev/loop9       30M   30M     0 100% /snap/snapd/8542
tmpfs           193M     0  193M   0% /run/user/997
tmpfs           193M     0  193M   0% /run/user/1000
```

Log2Ram fails to start:

```
Jul 17 12:24:53 minivm log2ram[640]: ERROR: RAM disk for /var/hdd.log/ too small. Can't sync.
Jul 17 12:24:53 minivm log2ram[663]: /usr/local/bin/log2ram: 61: mail: not found
Jul 17 12:24:53 minivm log2ram[662]: /usr/local/bin/log2ram: 61: echo: echo: I/O error
Jul 17 12:24:53 minivm kernel: [   17.257911] systemd[1]: log2ram.service: Main process exited, code=exited, status=1/FAILURE
Jul 17 12:24:53 minivm kernel: [   17.258009] systemd[1]: log2ram.service: Failed with result 'exit-code'.
```

0xf0ld, 2020-07-18:
Refer to https://github.com/azlux/log2ram/issues/90

kcalmond, 2020-07-20:
Also seeing same exact error/failure to start. Just upgraded pihole to 5.1 in case this is relevant. 
```
pi@cranberry:/var/log $ systemctl status log2ram.service
● log2ram.service - Log2Ram
   Loaded: loaded (/etc/systemd/system/log2ram.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2020-07-20 12:47:27 PDT; 13s ago
  Process: 14389 ExecStart=/usr/local/bin/log2ram start (code=exited, status=1/FAILURE)
 Main PID: 14389 (code=exited, status=1/FAILURE)

Jul 20 12:47:27 cranberry.almond.local systemd[1]: Starting Log2Ram...
Jul 20 12:47:27 cranberry.almond.local log2ram[14389]: ERROR: RAM disk for /var/hdd.log/ too small. Can't sync.
Jul 20 12:47:27 cranberry.almond.local log2ram[14389]: /usr/local/bin/log2ram: 61: /usr/local/bin/log2ram: mail: not found
Jul 20 12:47:27 cranberry.almond.local systemd[1]: log2ram.service: Main process exited, code=exited, status=1/FAILURE
Jul 20 12:47:27 cranberry.almond.local systemd[1]: log2ram.service: Failed with result 'exit-code'.
Jul 20 12:47:27 cranberry.almond.local systemd[1]: Failed to start Log2Ram.
```

azlux, 2020-07-20:
2 Errors in the same time. You don't have the basic command mail. But more important, `/var/log` is bigger than the SIZE value in the configuration file.
https://github.com/azlux/log2ram/issues/90#issuecomment-529399781

kcalmond, 2020-07-20:
moving to #90 thread

closed 2020-07-20 by azlux