azlux/log2ram #119: Log2RAM now doesn't want to install and uninstall

T-Shilov, 2020-07-21:
Hello, azlux,

First, I set Log2RAM with the command:

`apt apt install log2ram`

It installed and worked successfully.
Then I temporarily removed with the command:

`chmod +x /usr/local/bin/uninstall-log2ram.sh &&  /usr/local/bin/uninstall-log2ram.sh`

A few days later I tried again to install it with the command:

`apt apt install log2ram`

but got a message that it is already installed.
But this is erroneous, Log2RAM was previously deleted.

Now how do I set Log2RAM again and get it running?

Debian 10/64 bit.

Best regards,

azlux, 2020-07-21:
Hi,
If you have install log2ram with `apt`, you should remove log2ram with `apt` too.
Use `apt remove lo2ram`

T-Shilov, 2020-07-21:
But Log2RAM now does not want to be removed by the **apt** command!
When uninstalling, an error occurs:

```
Failed to stop log2ram.service: Unit log2ram.service not loaded.
Failed to disable unit: Unit file log2ram.service does not exist.
```

Why did you specify this harmful command?

`chmod +x /usr/local/bin/uninstall-log2ram.sh &&  /usr/local/bin/uninstall-log2ram.sh`

She only confused everything and ruined everything. I cannot now delete log2RAM :(

azlux, 2020-07-21:
This script exist because of the manual install. and I didn't update the Readme.
But in your case log2ram didn't exist anymore BUT apt thing it exist.
You can do a `apt install --reinstall log2ram` then `apt remove --purge log2ram` to remove it definitely.

MegaV0lt, 2020-07-22:
@azlux 
Maybe is a good idea to add an check if it is installed via apt aor not in the uninstall.sh

For example:
```
if dpkg -l log2ram ; then
  apt remove log2ram
fi
```

closed 2020-07-22 by azlux

T-Shilov, 2020-07-22:
**azlux**, thanks a great! Me managed to run your wonderful program again.

T-Shilov, 2020-07-22:
However, no !!! The application has just been installed, but does not work!
It didn't even create a config /etc/log2ram.conf

Wacky behavior of such a simple application. 
It needs to be fixed!

azlux, 2020-07-22:
@T-Shilov It's weird apt didn't reinstall well the package.
In this case, you can follow the "Manual" install, https://github.com/azlux/log2ram#manually
This will put files in the good folder without apt. APT should not warn you about any missing files.

T-Shilov, 2020-07-22:
**azlux** , but I don't want to use this barbaric manual installation method. I want to use only install **deb** -**packages**.
Please modify your Debian application and package so that they do not create obscure problems at the time of uninstallation or installation.

azlux, 2020-07-22:
@T-Shilov I mean the uninstall script have remove some file `apt` expect to have. This have been fix into https://github.com/azlux/log2ram/commit/1af7db2b28899a5584c01bc2d0fa17f416aac965 and will be available into next release.

But in your case, you need the manual install to replace files in the good place, then `APT` will be able to reinstall/purge/remove.

T-Shilov, 2020-07-22:
I'm scared to run your install.sh :)

MegaV0lt, 2020-07-23:
@azlux 
Please also add a check in install.sh, to avoid installation when already installed via apt ;-)

azlux, 2020-07-23:
@T-Shilov instead of run the install script, you can manually run L8 to 11 et L15-16. No surprise this way.
https://github.com/azlux/log2ram/blob/master/install.sh#L8-L11

@MegaV0lt Indeed

T-Shilov, 2020-07-23:
I all the same took a chance and installed Log2RAM  manually :)
Now it works really.

Tell me, how often Log2RAM do writes logs from memory to disk?
And what happens to them when the computer is suddenly turned off?

T-Shilov, 2020-09-25:
This log2ram is complete crap!

I have set the value SIZE = 256M
Everything went well in the beginning. The system worked for August and September..
But yesterday the df -h command showed the log reached 256M, and today LInux cannot reboot.
I have to press Reset.

After that I deleted this stupid log2ram

commit 1af7db2b (2020-07-22): rsync by default #115 + fix #119