azlux/log2ram #121: Wrong path to systemctl in log2ram-daily.service

Sucuarana, 2020-08-12:
OS: Raspberry Pi OS Lite (based on Debian Buster).

Hi,
I'm getting the following error via syslog:

Aug 12 00:00:19 pi-hole systemd[1]: Started Daily Log2Ram writing activities.
Aug 12 00:00:19 pi-hole systemd[10020]: log2ram-daily.service: Failed to execute command: No such file or directory
Aug 12 00:00:19 pi-hole systemd[10020]: log2ram-daily.service: **Failed at step EXEC spawning /usr/bin/systemctl: No such file or directory**
Aug 12 00:00:19 pi-hole systemd[1]: log2ram-daily.service: Main process exited, code=exited, status=203/EXEC
Aug 12 00:00:19 pi-hole systemd[1]: log2ram-daily.service: Failed with result 'exit-code'.

According to the link below, by default, systemctl is in "/bin/systemctl" in Debian/Ubuntu based distros.
https://serverfault.com/a/978288

glocalglocal, 2020-09-08:
OS: Raspberry Pi OS (based on Debian Buster).

As above, the following don't exist and I am not sure logs are ever saved. 
```
/etc/systemd/system/log2ram-daily.service
/etc/systemd/system/log2ram-daily.timer
```
I reinstalled log2ram but nothing changed.

Sucuarana, 2020-09-08:
> OS: Raspberry Pi OS (based on Debian Buster).
> 
> As above, the following don't exist and I am not sure logs are ever saved.
> 
> ```
> /etc/systemd/system/log2ram-daily.service
> /etc/systemd/system/log2ram-daily.timer
> ```
> 
> I reinstalled log2ram but nothing changed.

Hello,

Let me see if I understand, your installation does not have these 02 files?

`/etc/systemd/system/log2ram-daily.service
/etc/systemd/system/log2ram-daily.timer`

glocalglocal, 2020-09-08:
> Let me see if I understand, your installation does not have these 02 files?

Correct.

azlux, 2020-09-09:
don't having the service and timer, and having the bad linux systemctl path are 2 different issues.

For the path, it's weird, the debian image provide a simlink `/bin` -> `/usr/bin`, maybe raspbian decide to remove it.

For the files missing, did the installation give you any error ?

MegaV0lt, 2020-09-09:
> According to the link below, by default, systemctl is in "/bin/systemctl" in Debian/Ubuntu based distros.
> https://serverfault.com/a/978288

I think changing to `/bin` should work everywhere

glocalglocal, 2020-09-09:
> For the files missing, did the installation give you any error ?

No errors. I stopped log2ram and reinstalled, and I get: 

```
user:/etc/systemd/system $ sudo apt install log2ram
Reading package lists... Done
Building dependency tree       
Reading state information... Done
log2ram is already the newest version (1.5.2).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.

user:/etc/systemd/system $ ls -h log2*.*
log2ram.service

user:/etc/systemd/system/timers.target.wants $ ls -l
total 0
lrwxrwxrwx 1 root root 35 Aug 20 11:30 apt-daily.timer -> /lib/systemd/system/apt-daily.timer
lrwxrwxrwx 1 root root 43 Aug 20 11:30 apt-daily-upgrade.timer -> /lib/systemd/system/apt-daily-upgrade.timer
lrwxrwxrwx 1 root root 35 Aug 20 11:29 logrotate.timer -> /lib/systemd/system/logrotate.timer
lrwxrwxrwx 1 root root 32 Aug 20 11:30 man-db.timer -> /lib/systemd/system/man-db.timer
lrwxrwxrwx 1 root root 41 Aug 29 16:06 phpsessionclean.timer -> /lib/systemd/system/phpsessionclean.timer
```

Sucuarana, 2020-09-09:
> > According to the link below, by default, systemctl is in "/bin/systemctl" in Debian/Ubuntu based distros.
> > https://serverfault.com/a/978288
> 
> I think changing to `/bin` should work everywhere

I did just that when I reported. I edited the file "log2ram-daily.service" and changed the path to "/ bin / systemctl". Then I had no more problems.

Sucuarana, 2020-09-09:
> No errors. I stopped log2ram and reinstalled, and I get:
>
> ```
> user:/etc/systemd/system $ sudo apt install log2ram
> Reading package lists... Done
> Building dependency tree
> Reading state [...]

Remove the Log2ram that you installed from the repository and install it manually as described on the project's home page. I did my installation that way and all the files are present.

```
curl -Lo log2ram.tar.gz https://github.com/azlux/log2ram/archive/master.tar.gz
tar xf log2ram.tar.gz
cd log2ram-master
chmod +x install.sh && sudo ./install.sh
cd ..
rm -r log2ram-master
```

glocalglocal, 2020-09-09:
> Remove the Log2ram that you installed from the repository and install it manually 

This worked and now all files are in place. I even get a list of log files every time I start/stop log2ram. First time I see this since I installed log2ram a week ago. So, automatic installation doesn't work (for me).

azlux, 2020-09-09:
@glocalglocal indeed, the timer has not been release into the stable version. 
I think I will do it in the next 2 week.

glocalglocal, 2020-09-09:
@azlux until it is released, I think the apt method shouldn't be the recommended method, or at least there should be a note explaining that the timer-related instructions are not relevant.

closed 2020-09-09 by Sucuarana