azlux/log2ram #122: I install, reboot but it doesn't mount log2ram

fparrav, 2020-08-19:
I have installed using both methods (apt and manual) and I can't get it to work. I have a couple of network drives mounted in fstab, I removed and reinstalled them and it didn't solve the problem either.I am using raspbian buster.

```
pi@rpi1:~ $ df -h
Filesystem           Size  Used Avail Use% Mounted on
/dev/root             15G  8.7G  5.0G  64% /
devtmpfs             431M     0  431M   0% /dev
tmpfs                463M     0  463M   0% /dev/shm
tmpfs                463M  7.1M  456M   2% /run
tmpfs                5.0M  4.0K  5.0M   1% /run/lock
tmpfs                463M     0  463M   0% /sys/fs/cgroup
/dev/mmcblk0p1        43M   24M   19M  56% /boot
//10.0.0.201/Public  1.8T  1.1T  740G  60% /home/pi/WDMyCloud/Public
//10.0.0.201/Felipe  1.8T  1.1T  740G  60% /home/pi/WDMyCloud/Felipe
tmpfs                 93M     0   93M   0% /run/user/1000

```

mafioso12dk, 2020-08-20:
I have the same problem. Idk because it installs without any error

MegaV0lt, 2020-08-20:
Propably yput log is such big?
whats the outpit of 
`du -h /var/log`

fparrav, 2020-08-20:
Indeed the size of my / var / log was greater than 80 MB, sorry for opening a duplicate issue. Changing the `SIZE` value of in `/etc/log2ram.conf` fix it the problem.

closed 2020-08-20 by fparrav